synthetos/g2 #442: (Not so small) Moves are ignored due to code to avoid rounding errors

mhlong10, 2019-12-24:
It seems moves with a wide difference between major and minor axis travel and slow feed rates cause the minor axis to send no steps.  Example : Start at X0Y0 a G1X10Y.1F100 or G1X10Y1F10 will not step the axis at all (my machine is running X & Y at 100 steps per mm and Z at 320 steps per mm).

I think this boils down to the fact that the truncation is being done per segment, and segments are fixed in time, thus a long major axis travel time results in many segments, thus move per segment on the minor axis is below the threshold and gets completely ignored.

It looks like the following code was put in about 3 years ago when a lot of other changes came in.  **Does anyone know why the value of 0.01 was used rather than EPSILON to truncate?**  It is difficult to follow the precision of the math thru all the translations going on.

In _exec_aline_segment():

    kn_inverse_kinematics(mr->gm.target, mr->target_steps); // now determine the target steps...

    for (uint8_t m=0; m<MOTORS; m++) {                      // and compute the distances to be traveled
        travel_steps[m] = mr->target_steps[m] - mr->position_steps[m];
        if (fabs(travel_steps[m]) < 0.01) {                 // truncate very small moves to deal with rounding errors
            travel_steps[m] = 0;
        }
    }

If I change **if (fabs(travel_steps[m]) < 0.01)** to **if (fp_ZERO(fabs(travel_steps[m])))** it now steps as expected however I'm concerned about the rounding error mentioned.

justinclift, 2019-12-24:
Guessing you've also tried replacing the `0.01` with `EPSILON`? :smile:

mhlong10, 2020-01-05:
I haven't heard back from anyone about if the choice to use 0.01 rather than EPSILON was a calculated decision or just an arbitrary choice.

I have been running my machine with this line changed to  **if (fp_ZERO(fabs(travel_steps[m])))** and it has been running fine, however it is difficult to know if I'm accumulating small errors.  I also haven't done a long duration run yet.

- Does anyone know a good way to test for accumulated errors?
- Is there gcode available which would easily bring out accumulated errors?

justinclift, 2020-01-05:
> Is there gcode available which would easily bring out accumulated errors?

Do you have a rotary axis handy?  That could make for something suitable for generating G-code with testable accumulation errors.  eg something along the lines of wrapping around a rotation axis a few times (3600 degress or so)?

a-downing, 2020-01-07:
I think this should have been taken care of by following_error, but after looking at the line at stepper.cpp:708 it doesn't look like it would. `correction_steps = min3(correction_steps, fabs(travel_steps[motor]), STEP_CORRECTION_MAX);`. In the case where travel_steps gets set to zero, when less than 0.01, then correction_steps would also get set to zero. I see some other problems too. I think this should be done a different way. A segments starting position is set to the previous segment's target, when it should probably be set to the position that the previous segment *actually* stepped to, which isn't it's target if travel_steps gets truncated to zero.

justinclift, 2020-01-07:
@a-downing Any interest in writing a patch? :smile:

a-downing, 2020-01-07:
I was planning on playing around with it. My first instinct is to just get rid of the < 0.01 check and error compensation and see what happens. Also maybe getting rid of all the other approximate zero checks etc. Has this stuff been added to fix a problem, if so, what exactly was it?

`st_pre.mot[motor].substep_increment = round(fabs(travel_steps[motor] * DDA_SUBSTEPS));`
If travel_steps is very small, substep_increment still ends up being pretty large because DDA_SUBSTEPS is large. Did the < 0.01 check have to do with this?

justinclift, 2020-01-07:
Cool.  Just spoke to @giseburt quickly, and he mentioned it should be fine to remove the test. :smile:

mhlong10, 2020-01-09:
@a-downing 

> A segments starting position is set to the previous segment's target, when it should probably be set to the position that the previous segment actually stepped to, which isn't it's target if travel_steps gets truncated to zero.

I'm going thru some of this area of code in detail and questioned this as well.  It seems some of this code can be simplified too.  I'm going to try changing this and see if it fixes something I've noticed (see **side note** at bottom of comment).

@justinclift 

> Cool. Just spoke to @giseburt quickly, and he mentioned it should be fine to remove the test. 😄

I agree.  It seems this code in st_prep_line does effectively the same thing but using fp_ZERO.

```
        // Skip this motor if there are no new steps. Leave all other values intact.
        if (fp_ZERO(travel_steps[motor])) {
            st_pre.mot[motor].substep_increment = 0;        // substep increment also acts as a motor flag
            continue;
        }

```

**On a side note** - In looking at this in detail, I have noticed that simply stepping out to a position and back to the original position the encoder steps don't always go back to their original value.  Most notably, right after hard reset the machine is at zero, encoder at 0, step out to 10mm and it appears the encoder is +1 over what it should be, step back to zero and now encoder is at 1 and not 0.  I did notice that if I step out to 10mm and back again that encoder goes to +1 over again and consistently comes back to 1 so the relative steps on subsequent moves are okay but the absolute position is +1 over now.  I don't think this would be noticeable on a physical machine since the moves to home a machine will cause this initial absolute error, and the error doesn't seem to accumulate, but it does mean something isn't quite right...  which might just be what you pointed out.

The appears to be orthogonal to this thread's root issue.

a-downing, 2020-01-09:
I'm not so sure that fp_ZERO if statement should even be there. It seems like some of these things that look like they're intended to prevent rounding errors would actually increase them. If travel_steps[motor] is just under 0.00001 (what fp_ZERO checks for) then  `st_pre.mot[motor].substep_increment = round(fabs(travel_steps[motor] * DDA_SUBSTEPS));` would set substep_increment to around 85. It seems the job of substep_increment is to add to substep_accumulator for keeping track of when fractional steps add up to a full step. So just let it add 85 to it and do it's job? I might be misunderstanding something, but truncating 85 to 0 would just add to the errors.

a-downing, 2020-01-09:
Αctually, I guess it is needed because of the fabs() in `st_pre.mot[motor].substep_increment = round(fabs(travel_steps[motor] * DDA_SUBSTEPS));`. If due to rounding errors travel_steps was slightly less than zero, it would contribute to error in the same direction as if it were slightly above zero. Without the fabs() the positive and negative rounding errors would average out, but other code would need to be changed.

a-downing, 2020-01-09:
Ok, things are starting to make a little more sense. It seems like that `if (fabs(travel_steps[m]) < 0.01)` code in plan_exec.cpp is to prevent the `if (travel_steps[motor] >= 0)` code in stepper.cpp:678 from reversing direction if travel_steps is ~= 0, but changing signs. I think a better way to deal with this is to change `substep_increment` from an unsigned to signed int, remove the code that tries to account for rounding errors, remove the code that tries to determine the direction, and remove the `fabs` on `travel_steps` in `st_prep_line`. If `travel_steps` really is ~= 0, the errors should average out. After a step, `substep_accumulator` should be set to zero, so that `substep_increment` increments it in the positive or negative direction. That way when the absolute value of `substep_accumulator` adds up to a full step, the direction to step can be determined from it's sign instead of trying to determine the direction from the sign of individual fractional steps.

giseburt, 2020-01-09:
If you look at the four-cable-kinematics branch, you’ll see the code in the upcoming edge push (other changes haven’t been pushed):

The stepping code:
https://github.com/synthetos/g2/blob/four-cable-kinematics/g2core/stepper.cpp#L267-L356

And the loader code:
https://github.com/synthetos/g2/blob/four-cable-kinematics/g2core/stepper.cpp#L496-L615

Have changed to handle accumulation better, as well as to allow linear acceleration per segment.

And `_exec_aline_segment` has been changed to rely on the kinematics to set the steps directly instead of as a relative move, so there should no longer be an accumulation to be concerned with. The verbose inverse_kinematics call is translated [here](https://github.com/synthetos/g2/blob/four-cable-kinematics/g2core/kinematics.h#L54-L56) to just call `kn-> inverse_kinematics(target, steps)`, implemented [here](https://github.com/synthetos/g2/blob/four-cable-kinematics/g2core/kinematics.cpp#L126-L136).

I believe that those changes, along with providing better control int he kinematics layer and linear-acceleration in each segment, will prevent any accumulated error. But, I’d love to know sooner than later if I’m wrong. 😄

mhlong10, 2020-01-18:
I think we have had some good conversations here around and area which is about to have significant changes.  I've learned a lot about this area of code and I like the direction it is going.

I'm okay with the simple change I've made for now since it gets me past using an axis to clock data out of a laser raster buffer accurately.

I'm closing this for now.  We can revisit after the upcoming edge push.

closed 2020-01-18 by mhlong10