synthetos/g2 #443: Few wrong motor direction happening at random

maroussel, 2020-01-09:
Context:
I built an OpenBuild Workbee type Machine
I use an Arduino Due + g2Core to directly (without any shield)
it drive four(4) external stepper motor Drivers (DQ542MA)
I'm using the latest g2Core from Edge branch freshly rebuilt from yesterday

Workflow
-I'm using Fusion 360 to design and generate the G-code
-Using CNCJS(latest stable) on a raspberry Pi hooked to the Arduino Due that run g2Core

Made my own config for the workbee (I will share it as soon as the machine is stable)

Everything is working really nicely except for one major issue that is driving me crazy.
While moving the motors, they would sometime go in the wrong direction! (g2Core think they go in the good direction but they don't physically)

This happen completely at random from what I can see.

I can reproduce it by jogging the machine manually if I have a lot of patience, or better, when I engrave something, it happen sooner or later(within a minute or two), when the head moving during engraving and then I can "debug"
Let say I see the machine is not going in the proper direction, I stop it there and I send "G0 X1 Y1"(go in a given direction), what the machine actually do is the equivalent of "G0 X-1 Y-1"(opposite direction) as long as I do "G0 X1 Y1", "G0 X1" or "G0 Y1" it will go in the wrong direction "physically", g2Core think everything is going well, even if I wait for a very long time... but as soon as I send this "G0 X-1 Y-1"(Go in the opposite direction) then try again "G0 X1 Y1"(to in the original direction) , it will now go in the proper direction again and the bug is not there anymore (but the tool is not were the machine think he is and the engraving is completely wrong)

Today I was able to reproduce on 2 axis at the same time (x and y) and I had a tester for the directions pins. It was reading "High" when it was supposed to read "low". It didn't changed until I asked him to actually go in te direction it was really supposed to go on "High" then "low" again was fine.

Look like it's sometime not doing the change of direction on the output directly on the board, but mark it as changed in the variables.

I cannot say if the opposite happen yet where it read "low" when it should read "high".

I don't understand that I seem to be the only one with this issue either.

My next step will probably to dive into the code, but it will most likely be a long and slow process so any help or idea from more experienced people is welcome :)

justinclift, 2020-01-09:
Oh... looking at the DQ542MA photo here:

&nbsp; &nbsp; http://www.wantmotor.com/product/dq542ma.html

It's possible either the external dip switches, or internal jumpers might need changing around.

I've been using a (low end) external driver recently (Leadshine M542) that looks physically very similar on the outside.  Mine can run in either PULSE/DIRECTION mode, or CW/CCW mode, changeable via two jumpers (J1, J3) on the inside of the driver.  eg use a screwdriver to open it up.

Hopefully this is just a case of needing to change some switch settings. :smile:

ril3y, 2020-01-09:
That board wants 5v pulses. Does the due output that?  Also try pulling the pulse and dir + high. Then set step and dir active low on your config. Rebuild and flash. Try that.

mhlong10, 2020-01-09:
If you are driving the M542 directly with the DUE then the issue is most likely that it is not being driven sufficiently.  Being on the edge of the drive current required for the controller, the signals won't be interpreted consistently correct.  The DUE IO is 3.3v which is not enough to get the current needed across the resistor/diode combination in the controller.  And if you are driving in an open collector mode to 5v then, in addition to being in jeopardy of destroying the DUE, this also will work marginally because the drive current on the DUE is only a few milliamps.  Try using a simple circuit like Justin linked to.

maroussel, 2020-01-09:
First off, I'm new here and I wasn't expecting that much answers and that fast!! 
Thanks everyone for jumping in!

@JustingCliff 
- I didn't knew about the DUE debugger, at that price tag it's really excellent. I'll grab one.
- I'll defenetively give Visual Code a try now that I know I will need to spend some time in it.
- Sounded like electrcical issue to me as well, I looked at this part over and over again, but someone lended me a logic analyser yesterday and I was able to read that the Direction Pin was "High" (when it should have been "low") and not moving from "High" at all on the DUE. 
 Now I wonder what could cause a pin not to change on the DUE if it's not software, is it even possible? (My skills are software oriented, still learning everything on the electronic side)
- The dipswitch on the DQ542MA doesn't seem to be in cause here, they control: 
  1) Micro step resolution(1/16) 
  2) Standstill current (50% or 100%) 
  3) Max Output current (1 to 4.2 amp). 
I did put everything in a safer mode for now, even kept 100% power in standstill just in case. 
I might open the case later to see if there is any jumper, but for now I don't think the driver is in cause.

@ril3y & @mhlong10
- It's exactly what I was thinking too since I started this for the 3.3 vs 5, so I got bidirectional logic converter 5V-3.3V. 
  https://www.amazon.com/Logic-Converter-Bi-Directional-Module-Arduino/dp/B014MC1OAG
  This small circuit seem to work really nicely, but it work exactly the same without it as well. 
  DQ542MA seem to do fine with 3.3V signal logic already. Engraving is perfect in both case, no missing steps or anything, only the wrong directions from time to time in both cases. 
  So it seem pretty useless that I did add this logic converter now.

@ril3y
I'm not sure to understand that test:
"Also try pulling the pulse and dir + high. Then set step and dir active low on your config. Rebuild and flash. Try that."
You want me to wire this to active on active "low" instead of activate step/dir on "high"? (and configure the due according)
Never add a single issue on the steps pulse (both at 3.3V and 5V logic) and I can read the pin on the DUE is still "high" on DIR where it should be "low" when I experience the issue.

@mhlong10
I carefully keep 3.3V and 5V separate, I don't want to fry my only DUE board :) I don't understand 100% what you said about the "open collector mode to 5v", I'll do my homework to understand everything and I'll get back.

justinclift, 2020-01-09:
Looking around online for other people trying to connect external stepper motor drivers to the Due, everyone seems to hit the same problem. :frowning: 

As an alternative, these are the stepper motor drivers I used to use with my Arduino Due, when I converted my Shapeoko 3 to g2core:

&nbsp; &nbsp; https://www.pololu.com/product/1182

They can't pass as much current as the DQ542MA's, but they do work with the 3.3v signal from the Due and they're fairly cheap to experiment with. :smile:

justinclift, 2020-01-09:
Oops, didn't see your latest comment. :wink:

maroussel, 2020-01-09:
I guess you don't experiment any issue with the kind of "memory effect" on the direction pin ? From what I can see, it happen only when issuing a G0 command.

ril3y, 2020-01-09:
I am not getting this.  If you use a gShield with a due and g2 it works.  The direction pin has to be doing the right thing for the stepper drivers on the gShield.  @justinclift what other documented cases have you seen experiencing this issue with external drivers?

mhlong10, 2020-01-09:
@maroussel  Those logic converters will protect the 3.3v DUE from the 5v but it won't help the logic low drive current from the DUE chip.  It is still the DUE chip which needs to sink that current.

> Sounded like electrcical issue to me as well, I looked at this part over and over again, but someone lended me a logic analyser yesterday and I was able to read that the Direction Pin was "High" (when it should have been "low") and not moving from "High" at all on the DUE.

A big reason why this would be happening is if the DUE can't sink enough current.  It won't be able to pull the signal low enough.  Thus it will be interpreted as high by the analyzer because it can't pull the pin below the analyzer's low level threshold.

Try putting the DIR pin back into the condition where you are expecting it to be low.  Measure the voltage _between_ the + and - DIR pins directly at the controller.  If this isn't very close to 4.5 to 5v then the DUE is unable to pull the pin low enough (this assumes the power supply is supplying 5v... which is also worth measuring).

You could also measure the voltage across those pins when you are expecting it to be high.  You should see 0v.

It might not be the above but it's worth checking.

If you still have the analyze you could also try checking the DUE pin when it isn't connected to anything.

maroussel, 2020-01-09:
@mhlong10 Yes, I should be able to measure all those tonight ! One thing, I'm powering the DUE on USB only, do you think it would help to use an external power source for the DUE ?

justinclift, 2020-01-09:
> @justinclift what other documented cases have you seen experiencing this issue with external drivers?

I was meaning that when searching online for "stepper motor drivers" and "arduino due", there are quite a few people finding out the hard way that 3.3v from the Due doesn't work well with the 5v control circuit voltage expected by most larger sized stepper drivers. :wink:

mhlong10, 2020-01-10:
@maroussel If the 5v power is coming from the DUE and it is power by the USB, this will be a big issue too.  USB usually has protection diodes which drop the voltage a bit from about 5v to 4.5 or less.  It also can't provide much current and it sounds like you are powering that from a Raspberry PI USB port.  So one thing that can definitely happen is the Raspberry PI can droop the power supply depending on what the software is doing.

Try at least powering the DUE using a separate supply.  Even if this solves your problem for now, you are still running on the edge of specs.  So if you want a reliable machine I would definitely use some type of driver between the DUE and the controllers.

maroussel, 2020-01-10:
Ok I ran some test I'm more confused now...

Long version(Optional):
Ran with and external power, found a 12V/2A working (my 9v are to low in amp or wrong polarity)
I was able to run my most complex design to date (not really complet only2-3 min long) successfully, never happen in the past... I was about to say open the champagne bottle, but when I tried a second time, it fail right at the start.. ok, maybe an error from my side, tried a third time worked again... then fourth and it failed mostly at the beginning...
All in all, seem better (Or I was lucky) but defenetively not THE solution.

Short version: 12V power supply didn't work but seem to help

Then I tested Voltage at DIR+ and Dir- pin on the stepper motor...
Normal Low --> 0.017
Normal High --> 2.538V !!
When I have a Problem (Should be "low" but it's actually "High") --> 2.538V again

Tested the logic converter board 3.3V-5V, alimentation is pretty spot on, but at the logic pin, the 3.3V is good but the other side I was expecting 5V, but is actually at 2.538V !!, like the DIR+/- ?!? I don't count how many time I re-checked everything was wired properly and it was not my mistake! (Plus the info on board and on amazon)
https://www.amazon.ca/-/en/gp/product/B0148BLZGE/
I guess I would do better without it now?

This itself is really bad but I don't understand why it would prevent the Due from pulling the Dir pin Low?

@mhlong10 could you point me to an example for the "some type of driver between the DUE and the controllers." I was under the impression the logic board I got was this, but it look like it's not what I expected.

MitchBradley, 2020-01-10:
I would be very tempted to use a MOSFET between the DUE and controllers.  I will draw a schematic and attach it.

maroussel, 2020-01-10:
Thanks @MitchBradley !
I just did a test without the logic board but with 12V power and got the error really quickly
dir+/- pin
High 2.93V
Low 0.039V
High that should be low 2.93V
same pattern, but at little higher voltage

MitchBradley, 2020-01-10:
Was the driver connected?  Can you duplicate the problem with the driver disconnected?

MitchBradley, 2020-01-10:
Here is the schematic for using a transistor - either a FET or bipolar - to buffer between the DUE and the driver.  It inverts the logic sense so you need to configure g2core for active low.

[image]

maroussel, 2020-01-10:
It will be really hard to tell if I disconnect the driver as I rely on the position of the tool to know if its the wrong direction

MitchBradley, 2020-01-10:
Can you draw a diagram of exactly how you have hooked up the DIR+ and DIR- pins on the driver, including which DUE pin is connected?  If you don't have a schematic editor or a drawing program, just draw on a piece of paper and take a photo.

justinclift, 2020-01-10:
As a general concept, photo's are helpful too for people to get an idea of how things are set up.  If you're up for taking some. :smile:

maroussel, 2020-01-10:
Here some basic photos from some time ago... improved/changed a little sinsince but give a good overall setup Idea.
RaspberryPi on left with touch screen running CNC.JS and electronic "hidden" under the machine on a panel between both right legs of the Torsion table
[image]
The said Panel
column from left to right:
1)
- terminals for buttons, hooking relay, emergency stop... 
- Relay board to automatically start dust collector, router, power on the 24 & 36V
-DUE board (now provided with 12V 2.5A Power supply)
2)
-Main relay for the main power button
-24V Power Supply for accessories (router light, buttons led...)
-36V Power Supply for motors
3)
4 DQ542MA drivers--> X (M1), Y1(M2), Y2(M3), Z(M4)
4)
-terminals mostly for hooking the 6 limit switchs(2 on each axis)
[image]
I did made the board like a swiss cheese and behind I tried to put datacable together 
[image]
it's not on the picture since it's an older one, but it's on that side that I also put the temporary breadboard to test out the 3.3V to 5V breadboard converter
The schematics should be like that:
[image]

With 3.3V logic, I used the default g2Core pinout like the schematic on github
Dir/Step/Enable pin were connected directly from DUE to DIR+/Step+/Ena+ on the DQ542MA and the DIR-/Step-/Ena- are connected on the DUE ground.

With the logic converter shown above, The 3.3V & 5.5V & Ground are hooked to the DUE  3.3V & 5.5V & Ground. The signals pin on the LOW(3.3V) side go to the DUE and High Side(5V) go to the DQ542MA.

That lead me to two mains questions: 
1) What is the difference between the logic board I used and the driver schematics?
2) The main issue will still be the logic pin keeping the High value on the DUE(not moving to low)... will most likely still have the issue I guess, just a clearer High or Low?

Note: I can provide some drawing later if needed

maroussel, 2020-01-10:
One more detail... There are two motors on the Y axis (M2 and M3 on DUE), they are both on different pins in the DUE, but when I have the issue where the motors go in the wrong direction because of the "memory" effect (Pins were supposed to go on LOW but stay on HIGH), BOTH pins read HIGH, I never had a single desync (de gantry would be completely out of wack). I never had to realign the gantry.

mhlong10, 2020-01-10:
First off, nice looking setup.  Very clean.

I'm going to try to explain #1 with some crude cut and paste from previous diagrams (see below).

The current needed to drive the stepper driver flows via the green arrows.  You can see that in the suggested driver circuit the current flows straight thru the fet or transistor to ground and it takes very little current to turn on the fet/transistor (the red arrows).  However with the logic board the current still must flow to ground thru the DUE and thus the DUE must have the capacity to do this.  Additionally, there will be a slight voltage drop as it flows thru the logic board and if the DUE is asked to sink more current than specified, it will also have additional voltage loss.  All this loss adds up to not driving the stepper driver hard enough.

As a side note - the logic level will be inverted by the driver circuit but not thru the logic conversion board.  So your setting in g2core would have to change.  This isn't a cause of the problem here but wanted to let you know.  

[image]

maroussel, 2020-01-10:
Ok, Thanks @mhlong10 for the explanation and good comments on my setup and @MitchBradley for the schematics. Now I need to source the parts needed and try this out.
@mhlong10 was it normal that I was reading 2.538V between Dir+ and Dir- when High ? I was more expecting around 5V.

mhlong10, 2020-01-10:
@maroussel You should be seeing around 5v (or whatever you 5v supply is measuring).  As per my anser for #2, subtract that 1.0v and potentially 0.5v for the logic board loss (assumption here as I don't know the exact fet spec) then it would be 5v - 1.5v = 3.5v.  This is if you are measuring between the DIR + and - pins when driving the DUE pin low (or trying to).

MitchBradley, 2020-01-10:
I agree with what @mhlong10 said.  Those 3v3 to 5v bidirectional converters are worse than useless in this setup.  They are designed for high impedance (very low current) circuits like 3v3 microprocessors talking to 5v peripheral chips - not for driving optocouplers that need about 10 mA of current.

There is a reasonable chance that a circuit like this would work:

[image]

The analysis:

- Optocouplers typically require LED operating currents of about 10 mA.  They will often work reasonably well with 5 mA.
- The DQ542MA datasheet is unclear about the internal resistor - one diagram shows 200 ohms, the other 270.
- Optocoupler LEDs typically have a voltage drop of about 1.1V.
- With 3.3V supplied to DIR+ and DIR- at ground, the LED current would be (3.3 - 1.1)/200 = 11 mA or (3.3 - 1.1)/270 = 8 mA, depending on the actual value of the internal resistor.  Either is very likely to work.
- The ["standard" g2core pinout](https://github.com/cmcgrath5035/G2core-DUE-External-Interfaces/blob/master/Wiki%20Work/DUE%20tinyG2%20pinout%200.2.pdf) uses "high current" DUE pins for M1,M2,M3,M4  STEP,DIR,ENA signals EXCEPT for M1 STEP which is a low current pin.  High current DUE pins can source 15 mA and sink 9 mA.  In the circuit shown above, the pin is sourcing the current, so it is within specs (11 mA < 15 mA).
- The low current M1 STEP pin is a problem.  Low current pins can source 3 mA and sink 6 mA - not enough for an optocoupler in source mode and marginal in sink mode.  So that signal should be moved to a different pin that is high current.
- With 4 motors each with step/dir/ena, there are 12 pins driving optocouplers.  If the current is 8 mA/pin, that is about 100 mA total.  If it is 11 mA/pin, that is 132 mA total.  The [ATSAM3X datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf) says, in section 45.1, that the 144 pin LQFP package - which is what is on the DUE - can supply a total of 130 mA from its I/O pins.  The limitation is heat dissipation - all that current makes the chip get hot.  So you are on the hairy edge if the optocouplers have 200 ohm dropping resistors - but if so you could add a little external resistance in series and it would probably still work.
- But we haven't considered the relay board.  The optocouplers on those relay boards need current too.  Some relay boards have a driver chip so the microprocessor doesn't have to supply all the optocoupler current, but I don't see such a chip on your board, so we have 4 more optocouplers to supply at 10 mA each.  That's over budget under pretty much any calculation.

So it would be a good idea to put buffering between the DUE and the DQ542MAs.  One choice would be the MOSFET circuit as shown in a previous post.  You would need 16 MOSFETs, one for each optocoupler. Another would be to use two ULN2803 8-channel driver chips.

maroussel, 2020-01-10:
Thanks @MitchBradley. I will defenetively try this. I like the idea of the two ULN2803 if I can find it here. It might take some time to get the parts so don't hold your breath. But I'll post back as soon as I can test it out!

MitchBradley, 2020-01-10:
This is the no-compromises way to do it.  Digi-Key sells ULN2803A for $1.19 each.

[image]

To clean up the wiring and reduce the chance of bad connections, you could make a hand-wired shield for the Due.  Start with a 6cm x 8cm perfboard and solder in header pins for the Due connectors you need to mate to.  You could put pins everywhere but it will be hard to insert and remove.  Add 0.1" pitch screw terminal block connectors for the wiring to the stepper drivers.  Solder the ULN2803 chips to the board and wire them to the connectors.

Do yourself a favor and get a crimp ferrule kit like this - it will make connecting to screw terminals much easier and more reliable https://www.ebay.com/itm/113918126058

MitchBradley, 2020-01-10:
You can get ULN2803A from Amazon or eBay.  From China you can get 10 for $2.

I would be tempted to use pluggable terminal blocks for connecting to the drivers.  https://www.amazon.com/dp/B07B9R5BQH . That would make maintenance easier because you could connect the wires away from the board then just plug it, instead of having to screw wires down in a cramped space.

maroussel, 2020-01-10:
Thank you so much again! I'm always looking to learn as much as I can and I defenetively learned a lot in the last few days with all the help I got here. 
One more simple question, what should I use if I want to make similar diagrams?

MitchBradley, 2020-01-10:
I use KiCad.

If you decide to use pluggable terminal blocks, be sure to the the 5.08 mm (.2") pitch version.  The 3.81 mm (0.15") version will not fit in a standard 0.1" pitch perfboard.

mhlong10, 2020-01-11:
@maroussel  I'm glad to see you are going to add proper drivers between the DUE and the stepper driver.  You've done such a nice job on the rest of the machine so it is good to make this piece solid as well.

Even with this, please continue to use separate power and not USB power to supply the DUE when driving the machine.  This will give you more a more stable supply rail with higher current capability.

maroussel, 2020-01-11:
@mhlong10  excellent, I'll continue power it on the 2.1mm socket, but I was wondering if I should get a 9V one or if my 12V (2.5A) is fine? Do not want to risk overheating it. I read about that being possible over 12V, but didn't seen any excessive heat buildup my few tests.

mhlong10, 2020-01-11:
@maroussel I think the older DUE had issues with greater than 9v supply but the new ones, which have been around for quite some time, have a new switching regulator so using 12v isn't a problem.  I've been powering mine from a 12v supply and had no issues.

justinclift, 2020-01-11:
@MitchBradley @mhlong10 On a similar note, any idea if this Due shield - specifically made for g2core - has the needed pieces already built in?

https://www.djuke.nl/en/products?page=shop.product_details&flypage=flypage.tpl&product_id=406&category_id=71

My electronics skill at the moment is still extremely basic, so although I can see various things on the [schematics diagram](https://www.djuke.nl/components/com_virtuemart/shop_image/product/G2Shield-r2-schematic2.png) I'm not sure if they'll take care of the above problem. :wink:

MitchBradley, 2020-01-11:
Looks perfect to me.  The optocoupler driving circuit shown here
[image]
is the same as the NPN version of the circuit that I posted earlier, except for the resistor value.  Either resistor value will work and drive any modern transistor to the fully on state, without overloading the Due.
I like this board a lot.  The choice of 6-pin IDC connectors is a good one.  IDC connectors are easy to assemble to ribbon cable with just a vise.  In a "pinch", you can do it with pliers.  I have done it with a door.  You open the door, put the connector in the door jamb at the hinge side, and close the door until it squeezes the connector into the ribbon cable. You can buy the connectors and the cable from the same site, at reasonable prices.
The board also has RC filtering on the input pins - another good feature.
You will need a soldering iron.  The instructions say that the through-hole components - mostly the connectors - have to be soldered on by the user.  The surface mount components are already soldered.

MitchBradley, 2020-01-11:
He has a new webstore at https://webshop.djuke.nl/kits/g2shield-onboard-kit .  You can pay an extra 20 euros to have the board completely assembled.  The IDC connectors etc are not yet on the new store but he says contact him and he will move stuff over to the new store.

Maybe we should reach out to this guy and see if he wants to participate on the team at some level.  As a DIY audio guy myself, I looked at his audio projects and was very impressed.

justinclift, 2020-01-11:
Awesome. :smile:

We should probably add a link to the board - on his new store - from somewhere obvious in our wiki.  Maybe we should have a new wiki page along the lines of "Hardware known to work with g2core"?

Probably including this, the Due, the (when released) gQuintic, "known good" stepper drivers, etc.

justinclift, 2020-01-11:
Ahhh, this is his new web store page page for the kit for "external drivers":

&nbsp; &nbsp; https://webshop.djuke.nl/kits/g2shield-external-kit

Nicely, it even lists the DQ542MA that @maroussel is using.

@maroussel This might be your best approach. :smile:

maroussel, 2020-01-11:
Yeah, It would be my safest bet for sure for my setup, but on the other side, I really want to experiment and hone my skills so I continue on the path of building my own board for now. With the help I got here, and really good explanations, I'm pretty confident now ;)

MitchBradley, 2020-01-11:
In exchange for the help we require that you post photos of what you come up with!

maroussel, 2020-01-11:
I will for sure! Hopefully it will also help and inspire other to use G2 as well.

maroussel, 2020-01-15:
Some good news: 
I tried to do various tests mentionned here while I'm waiting for the parts.

unsuccessful:
- I unplugged the wire after I had the issue to see if the pin would change.
- I tried to find a way to test without wire, could not figure out how.
successful:
- I went back to basic with a bit safer config including 1/4 microsteps (1/16 previously) and no input 
I then flashed the config and unplugged the limit switch
Result: I was not able to reproduce any desync after that.

I ran at least one hour of tests without a single desync before I stopped. The previous record was about 4 min before a fail. (same gcode file)

So it look to me that the external motor controllers don't mind the 3.3V much, but I guess by removing the limit switch I was able to relieve some current on the DUE like it was mentionned earlier.

Now with the 5V driver hopefully I will be able to have both motors and limit switchs.

Note: I had an "issue" where it went into "hold" status for no apparent reason. My lack of knowledge here is probably more to blame.

MitchBradley, 2020-01-15:
Now I am wondering if the hold could have been related to false triggering of a limit switch due to noise.  It is often necessary to use RC filtering on limit switch inputs to eliminate noise spikes, especially in systems with high-current drivers.  Normally closed limit switches are better than normally open in this respect, because it is much harder for noise to override a low-impedance closed switch than the high impedance state of an open switch circuit.

maroussel, 2020-01-15:
Yes, but there was no limit switch connected on the Due, could it be noise even in that case ? 

In that regard to limit, I was thinking on going the same route as Menno de Graaf (Djuke.nl).

[image]

And maybe even waiting before building the motor driver to see if the problem I had was not actually releated to the limits? But I guess I might get back to hitting the current limit again if it was the issue.

MitchBradley, 2020-01-15:
Since you are using an Arduino Due directly, you really need external pullup resistors on any input pins that you use for switches.  The built-in pullups are nominally 100K, which is too high to provide significant noise immunity.   4.7K external pullup resistors would be good, with 0.1uF to ground to absorb any high frequency noise spikes that are coupled into the wiring from electromagnetic fields resulting from motor switching, spindles, etc.   4.7K resistor between 3V3 and the input pin, 0.1 uF capacitor between the input pin and ground, both located close (within a few cm) to the input pin.  These components can go on the board with the ULN2803A chips

MitchBradley, 2020-01-15:
So the "hold" was a separate issue?  I didn't quite understand that part.

10K is fine as a pullup as shown in the schematic you attached. It doesn't have to be 4.7K. I would not go higher than 15K for this purpose.  100nF is the same as 0.1uF.

I worry about overheating the CPU chip by directly driving all those optocouplers.  In my setup I am only directly driving 7 total optos, so I am well within the power limit, but your setup is pushing the limit pretty hard.

MitchBradley, 2020-01-15:
If you are short of board space, or just want a cleaner look, you can get resistor packs that are pre-bussed.  You connect one pin to 3v3 and the rest to inputs.  Here is one https://www.alliedelec.com/product/bourns/4608x-101-103lf/70154307/
Be sure to get the "bussed" version (Bourns 101) instead of the individual resistors version (102).

MitchBradley, 2020-01-15:
You can get them with different numbers of resistors.  Here's one with 6 resistors, the same number as in Menno's schematic https://www.alliedelec.com/product/bourns/4607x-101-103lf/70154607/

maroussel, 2020-01-15:
Yes, the Hold happened after some time testing, once I didn't got any desync on the axis and the Limit switch were already removed. I don't know what caused it, I tried to find, but I didn't wanted to spent too much time to figure it out since I wanted to concentrate on the main issue of the desync. I added it in case is there are common cause for g2Core going into "Hold". With the desync I was not able to build that much experience yet.

MitchBradley, 2020-01-15:
I have seen g2core go into hold when I exceeded a soft limit.  One of my programs was traveling just outside the configured soft limit area.  It was on an arc so the x and y values in the GCode line were within the machine envelope, but the arc path just barely clipped outside the limit.

maroussel, 2020-01-15:
Soft limit were disabled too. I begin to think maybe I should test with an original DUE. I'm using a clone for now. I was thinking to get an Italian made after my experimentations, maybe I'll swap it earlier.

justinclift, 2020-01-15:
It'd might be useful to get the Segger J-Link thing (the hardware debugger) as well.  That way can you put a breakpoint in the hold code and see what's happening (software wise) when it fires.

Especially if it turns out to be a weird software glitch instead of noise or similar. :smile:

MitchBradley, 2020-01-15:
"printf style" debugging works too.  You can just dump information to the serial port when it goes into hold.  You can't leave that in for production use but it is okay for looking at specific situations, and avoids the learning curve for hardware debugging UIs.  Hardware debugging is great if you are doing a lot of it, printf has a lower barrier to entry for one-off situations.

justinclift, 2020-01-15:
Good point. :smile:

maroussel, 2020-01-15:
@MitchBradley what do you think would be the best in my case, the ULN2803A drivers or the optocoupler driving circuit? 
I might also test, like you pointed earlier, to add a little external resistance in series for driving the motors for now.

MitchBradley, 2020-01-16:
There is very little functional difference.  The ULN2803N is essentially the same as the discrete driving circuit that I showed. It has the advantage of being more compact and easier to wire because there are fewer duplicated nodes - with the discretes you need to wire up all the grounds whereas the ULN2803A has one ground pin that is connected internally to all the transistors.  Also, compared to the discrete circuit with the NPN transistor,  you do not need the base resistor because that is inside the package.
The advantage of external resistance for motor driving would be to reduce the power consumption of the Due, at the expense of reducing the drive current available to the optocoupler.  That might help overall, or it might hurt.

maroussel, 2020-02-17:
Quick update, I got everyting needed to build the board, I rewired the machine for new driver setup, changed the config file onthe DUE, did a quick test with solderless proto board. Everything seem to work fine now, at last because I lost a lot of time figuring out why an issue I had originally. In the end I started all over again wiring + config and it worked first try after that.
This is the board used for the quick test:
[image]

Coming up next: making the perfboard with everything on it! (I already have a good idea of the Layout)

MitchBradley, 2020-02-17:
If you want to take a shot at doing a PCB, you might have a look at easyEDA. I tried it last week for a prototype I'm doing for a friend, and was extremely impressed at how easy it was to use.  I'm a longtime KiCAD user and I was able to start using easyEDA pretty much instantly. It is integrated with a PCB fab service "jlcpcb" that offered 5 boards for $2 (shipping is extra but reasonable)!  I completed my design, from schematic entry to placing the order including assembly in 2 days. Since your components are through-hole and you already have them, you probably would not want to have them do the assembly.  I placed the order a week ago and my PCBs are complete, but assembly has been delayed due to coronavirus.  Apparently their 2 layer PCB service is up and running but their other services are delayed.  If you design in easyEDA or KiCAD or any other system, you can have the boards produced anywhere.  For example, https://oshpark.com/ make boards in the US and ships anywhere in the world at no extra cost.  It's a bit more expensive than jlcpcb, but still quite affordable - $10 for 3 copies of a 2 square inch board, delivered.

MitchBradley, 2020-02-17:
Even if you choose to go the perfboard route, I often do a PCB design just to prove the layout.  Set the PCB editor grid to 100 mils to match the perfboard, place the components, and move them around to optimize the routing.  If you need guidance or get stuck, just ask.  We can connect on Facebook or Slack if you would like.

[27 more replies]

closed 2020-04-09 by ril3y