synthetos/g2 #447: X and Z axis motors keep turning after endstops are hit

Johnr24, 2020-01-24:
My max travel on my machine is 530 (X) and 319 (Y) 180 (z)
I find that my machine is continuing to turn the motors past the endstop until they have gone from -530 (home) to -1060 for the X and on the Z +180 from 0 (home)

(full config in my profile I'm using the shapeoko2dualY.h settings file)

justinclift, 2020-01-24:
No worries.  Is this with the `master` branch, the `edge` branch, or something else? :smile:

Johnr24, 2020-01-24:
> No worries. Is this with the `master` branch, the `edge` branch, or something else? 😄

Hi 
This is with the edge branch, all my config is in my profile) please do tell me if anything is out of place)

justinclift, 2020-01-24:
Cool.  I forgot to ask, which hardware are you running on?  eg is this an Arduino Due + stepper driver board of some sort + stepper drivers, or something else? (etc)

justinclift, 2020-01-24:
As a data point, the limit switch settings are this piece:

https://github.com/Johnr24/g2/blob/fe17519807e32b110c3743b1cddb95a58e9b12ff/g2core/settings/settings_shapeoko2dualY.h#L179-L211

It's a matter of making sure the "digitial input" pins are wired up correctly, and the settings are set to the correct value.  One common gotcha is not having the right type set (`NORMALLY_CLOSED` vs `NORMALLY_OPEN`), which will mean the signal is ignored.  If you have a multimeter handy and the connections for the limit switches are easy to access, it's fairly simple to test which type you have.  There are other ways to check though, if you don't have a multi-meter handy. :smile:

Johnr24, 2020-01-24:
> Cool. I forgot to ask, which hardware are you running on? eg is this an Arduino Due + stepper driver board of some sort + stepper drivers, or something else? (etc)

Yeah I'm using an ardino due with a some external stepper boards the TB6600s

Johnr24, 2020-01-24:
> It's a matter of making sure the "digital input" pins are wired up correctly, and the settings are set to the correct value.

I get the stat 3 code when all three endstops are pressed, 

I double checked all of them with the multimeter and I'm running normal closed on the end stops as the code also suggests!

Hope that helps

justinclift, 2020-01-24:
Cool.  A Due + directly wired stepper drivers is a valid combination, so we should be able to get this working. :smile:

With the wiring, are you using the pinouts from the PDF here?

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core

For compiling the code then flashing the Due, which commands are you using?

Oh... I also forgot to ask, which Operating System are you using? :smile:

While we're at it, which G-Code sending program are you using, so I have an idea how you're sending commands to the thing to trigger the homing operation?

Sorry for the 20 questions approach.  The extra info helps me understand the pieces you're using and (hopefully) see any potential pitfalls. :smile:

Johnr24, 2020-01-24:
I'm using, 
The standard pinouts from the PDF
For compiling I run 
make PLATFORM=DUE CONFIG=ShapeokoDual BOARD=gShield

I'm on OSX 10.14.5

I've tried using both UGS platform and CNCjs and I'm getting the same issue

Johnr24, 2020-01-24:
sorry for uploading I use the OSX TinyG2 OSX Programmer with the standard commands in the wiki

justinclift, 2020-01-24:
That's all good info. :smile:

> I get the stat 3 code when all three endstops are pressed,

That just sunk in.  It sounds like when individually testing each of the digital inputs (eg pressing the switch with your finger, then looking at the value of (say) `$di1`), you can see the state change properly.

That should mean things are physically wired up correctly at least.

So it sounds like a software or configuration problem of some sort.

Nothing obvious springs to mind as to what the problem might be, but I have a feeling I've seen this before and did turn out to be something simple. :face_with_head_bandage:

Hmmm, I have a Due here at the moment that's not doing anything.  I need to go out for a bit, but when I'm back I can flash the `edge` branch on to it and refresh my memory of how the limit switches work in practise. :smile:

justinclift, 2020-01-24:
Hmmm, I think I'm going to need my stuff from the Makerspace.  I'll head in there tomorrow to get everything, and try to replicate the problem using your g2 settings file. :smile:

Sorry for the delay.  Pretty sure this will just be something simple though. :smile:

Johnr24, 2020-01-24:
Thanks, if you could use my fork? that would be ideal!
it defiantly seems software,  The key thing is that regardless of where you start the homing procedure, it will always reverse the axis (towards the switch) by 530mm (so x-530) which also happens to be the axis length

justinclift, 2020-01-24:
Yep, will use your fork.  I'm poking around at it now, not that I have hardware (apart from the Due) that I can test on right atm. :wink:

justinclift, 2020-01-24:
Hmmm, with this:

```
make PLATFORM=DUE CONFIG=ShapeokoDual BOARD=gShield
```

Should that `CONFIG` value be `ShapeokoDualY` instead of `ShapeokoDual`? (with a `Y` on the end)

Only asking because I can see `ShapeokoDualY` listed in the [`boards.mk`](https://github.com/Johnr24/g2/blob/fe17519807e32b110c3743b1cddb95a58e9b12ff/g2core/boards.mk#L30) file, but nothing for `ShapeokoDual` (without the Y).  I *think* a miss there means the CONFIG option will be ignored, so it won't be picking up the correct settings file.

Johnr24, 2020-01-24:
I know it's picking up the settings file that I'm writing to?

justinclift, 2020-01-24:
No worries.  This part of the makefiles is kind of opaque to me personally. :wink:

justinclift, 2020-01-24:
As a thought with the axis values, it's (generally) more standard to have the `X_TRAVEL_MIN` be 0, with the `X_TRAVEL_MAX` be (say) 530.

Same thing with the Y axis (eg `Y_TRAVEL_MIN` 0 and `Y_TRAVEL_MAX` 319).

Not sure it'll make any difference for the homing, but it's closer to how the everyone else does it, which might help with us figuring out what's going wrong. :smile:

justinclift, 2020-01-24:
Oh, this is likely a problem:

https://github.com/Johnr24/g2/blob/fe17519807e32b110c3743b1cddb95a58e9b12ff/g2core/settings.h#L59

That should probably be changed back to `settings_default.h`, which is a special file used to fill out any default settings for things not mentioned in your specific settings file.

It'd also explain why it's finding your settings file with the slightly wrong command line above.

Um, try with the settings_default.h back in the above spot, and using the `CONFIG=ShapeokoDualY` piece on your make line.

I just tested it here, and compile wise things look a bit better.  Not get any weird warnings about pin redefinitions, which I was when trying with your `edge` branch. :wink:

Johnr24, 2020-01-24:
Thanks I'll change that now and check this over tomorrow!
If it works I'll let you know!

justinclift, 2020-01-24:
Cool. :smile:

Johnr24, 2020-01-25:
Hey!
I've tried that and I'm still getting the same issue!
I've changed a few things around including the boards.mk file to stop myself getting confused! It's all committed and pushed on my fork!

so I'm now configuring using this (it's just a 2 that's changed,
make PLATFORM=DUE CONFIG=Shapeoko2DualY BOARD=gShield

The Y axis is still fine! The x and z axis are still misbehaving

justinclift, 2020-01-25:
No worries.  I'm heading into the local Makerspace where most of my stuff is in a few hours, and I'll make sure to take the Due.  I should be able to set things up then very similarly to what you have, to try and replicate the problem.  Then figure out how to fix it. :smile:

Johnr24, 2020-01-26:
Morning, 
Did you make any progress on solving the problem? where you able to recreate the issue?
Best
J

justinclift, 2020-01-26:
Actually feeling really guilty here.  Got distracted with other stuff and didn't get to the Makerspace. :frowning:

I'm definitely heading there tomorrow though. :angel:

Johnr24, 2020-01-27:
That's okay! Any luck today 👏🙏🙌

justinclift, 2020-01-28:
And... just starting on it now.  Getting it done as a page on our wiki, as other people are asking for the same thing too.  Might as well kill multiple birds with a single stone.

https://github.com/synthetos/g2/wiki/Walk-through,-using-a-Due-and-lower-power-stepper-motor-drivers

(That page title may need shortening, not sure yet)

I have the initial bits needed, minus actual limit switches just yet.  There is an Xcarve clone in the local Makerspace that I've been getting up and running, which I can probably repurpose to this once I have the basics (Due, steppers, etc) all working correctly.

justinclift, 2020-01-28:
Gah.  Just realised using a breadboard to keep things as simple as possible probably isn't going to work.  Can't (safely) pass anywhere near enough current through a breadboard to feed the stepper motors.

May need to get some protoboard, unless I can think of a workaround. :wink:

How are you hooking everything together?

justinclift, 2020-01-28:
Whew, it turns out the Makerspace has some protoboard here. :smile:

Johnr24, 2020-01-28:
I just use lots of butchered and then wago blocked jumper cables, so I come straight off the arduino and into the the TB6600s ( I crimp the other end of some silicone wire (because it's more flexible and can take more current without melting) and stick that crimped end in the terminal block on the motor driver, this way I can also run one software axis for two motor drivers using a 3-way wago block

https://www.wago.com/gb/electrical-interconnections/discover-installation-terminal-blocks-and-connectors/221

Here's a link to the wago blocks, They take a ridiculous amount of current (41 Amp at 450v) - You can get them on amazon/RS/Misiumi/Ebay super useful

justinclift, 2020-01-28:
Cool.  That sounds *way* better than my approach here this time around.  Have gone with protoboard, jumper wire + hookup wire.  It's taking ages to solder. :frowning:

I think it's going to be a few days until I have this in a usable state, going by my current speed of progress.  Ugh.

justinclift, 2020-01-30:
Still getting this done. :wink:

The local Makerspace doesn't have any capacitors, so bought some today.  eg for filtering against voltage spikes on the stepper motor drivers.  

Then realised I'd put the wrong headers on the board.  Board version 2 coming up.  It's going faster the 2nd time around.

Johnr24, 2020-01-30:
Hi
Here is mine as it stands, 

[image]
I'm using one of those opto-isolation boards for my endstop filtration, I got mine from tindie, 
The TB6660s are underneath the router mounted to the aluminum extrusion

Johnr24, 2020-01-30:
Just an idea, 
How about a file for a PCB (gerbers)  on it that would work as a breakout board It would turn Arduino headers into wago clamp downs? really simple to design but would neat and tidy the whole thing up 

could go on that page you mentioned on the wiki, if their is interest I could knock something up

justinclift, 2020-02-01:
Killed my Due. :man_facepalming:

Finished the soldering last night, and tested each connection to make sure there weren't any shorts.

Hooking things up today, one stepper driver at a time then testing, the 1st and 2nd motor drivers went fine.  **Something** seems to have gone wrong with the 3rd though, and now the board seems to be fried.  Won't even respond over JTAG. :frowning_face:

First time I've ever killed a board.  Dammit.

Have ordered a clone board from Ebay... it'll take a few days to arrive.

In the mean time, lets try a different approach to solving your problem. :smile:

1. Instead of using your current (modified) source code, do a fresh clone of the g2 repository somewhere and we'll make some fresh changes in that.
2. We'll be changing (only) the `g2core/settings/settings_default.h` file, and compiling using `make PLATFORM=DUE BOARD=gShield`.  Just to start with "known good" defaults.
3. First, copy the individual motor and axis settings settings from your config file, overwriting the value in the settings_default.h file
   * https://github.com/Johnr24/g2/blob/31ce8a74d75ca814beb17687fb9656689f299351/g2core/settings/settings_shapeoko2dualY.h#L84-L157
4. Try compiling, flashing and moving the axis around with that, just to confirm things are working as expected.
5. If that works (it should), then change the digital input pin types from `IO_ACTIVE_LOW` to `IO_ACTIVE_HIGH` (to match your hardware) for DI's 1, 3, and 6.  **Don't** change the `ACTION` nor `FUNCTION` line for any of the digital inputs (yet) though.
  * You'll also need to update the "homing input" value for each axis, so X = 1, Y = 3, and Z = 6.
    * https://github.com/synthetos/g2/blob/c684edcd38a18e4014860a9bf1ace1dc3b37285f/g2core/settings/settings_default.h#L442
    * https://github.com/synthetos/g2/blob/c684edcd38a18e4014860a9bf1ace1dc3b37285f/g2core/settings/settings_default.h#L483
    * https://github.com/synthetos/g2/blob/c684edcd38a18e4014860a9bf1ace1dc3b37285f/g2core/settings/settings_default.h#L524
  * Possibly interestingly, the Z axis homing input value in [your current config file](https://github.com/Johnr24/g2/blob/31ce8a74d75ca814beb17687fb9656689f299351/g2core/settings/settings_shapeoko2dualY.h#L152) seems to be 5 rather than 6.  It'll be funny if this turns to be the problem. :smile:
  * Anyway compile, flash and test that.  Hopefully homing works properly now.  If not, at least this is on a foundation that shouldn't be too hard to figure out. :smile:

MitchBradley, 2020-02-01:
Note that you can test the pin configurations for the homing inputs with JSON commands, without having to recompile for each test.  Then when you are satisfied with the behavior, transfer those settings to the config file, recompile and download once.

justinclift, 2020-02-02:
It doesn't have to be JSON commands to reconfigure most settings on the fly.  But yeah, good point.  Settings can be changed "live" then tested, without needing to recompile. :wink:

Johnr24, 2020-02-02:
Hi, 
I've compiled that and flashed it, 
I've I copied the values into the new file rather than copying whole lines of code or anything weird,

however that isnt helping I'm still getting the same result of my Y-axis working fine :)
but the X and Z axis not working okay.

MitchBradley, 2020-02-02:
Have you tested to see if g2core is seeing the switch states properly?  You can read the input values by typing
`$in`
in a terminal window.  Manually activate the switches and see if the output from that changes.
Refer to this page to see what other commands are available.
https://github.com/synthetos/g2/wiki/Digital-IO
You can use either the JSON form `{di1:n}` or the equivalent command form `$di1` for ease of typing. To set a value you can use either the JSON form `{di1mo:0}` or the command form `$di1mo=0`

justinclift, 2020-02-03:
For typing stuff in manually (eg into the g2core console area in CNCjs or UGS), it can be helpful to temporarily set the output to text mode.  That's done by sending `$ej=0` (and hitting enter).  Text output is a lot easier to read, at least for me. :smile:

Johnr24, 2020-02-05:
Thanks for your help the $in command was instrumental in getting this working!
Wasn't sure what it was in the end, think it might been a connection problem or something? with the endstops to the board 
Can you tell me which pin  is on DI11 I'm wanting to get the emergency stop button i have working

Johnr24, 2020-02-05:
I've figured out where DI11 is it's the CMIN pin

closed 2020-02-05 by Johnr24

justinclift, 2020-02-05:
All working ok now?

Johnr24, 2020-02-06:
Yes, In terms of function
But I'm trying to get something cut and ocasionally it will Just complety stop working mid cut. super annoying! how do I start to fix this?

Johnr24, 2020-02-06:
X18.729 Y74.515 Z-12.911 - this was the line of code that it stalled on this time, however It has gone past this line completly okay before

Johnr24, 2020-02-06:
Just an update I just pressed cycle start biutton again in CNC-JS and it started up again?
Does the Due need active cooling, it's 10.4C in my shed at the moment

MitchBradley, 2020-02-06:
As it happens, the g2core developer group on slack is actively discussing/debugging a very similar - could be the same - stall problem right now.  Best guess at the moment is that it's a problem that Rob has already fixed, but the fix is in the edge-preview branch that has not yet been released.  Several of us are trying to accelerate the process of getting that deployed.  There is some work yet to be done to backport it to Due and other platforms.  The good news is that Top People are working on it :-)

ril3y, 2020-02-06:
Justin. Have we verified that this infrequent pausing is a due g2 issue?
Or does this occur on the quintics?

karoria, 2020-02-07:
> Just an update I just pressed cycle start biutton again in CNC-JS and it started up again?
> Does the Due need active cooling, it's 10.4C in my shed at the moment

I also faced such issues with grbl esp32 port and cncjs, where pressing cycle start makes the paused machine working. In my case it came out to be a noise issue though i was using shielded usb cable between rpi4 and grbl. I inserted ferrite cores at both ends of usb cable and the problem seems gone. This may be irrelevant to your case but i thought sharing this for your experimentation.

MitchBradley, 2020-02-07:
@ril3y - According to what Rob said on slack, there was a race condition where the step interrupt handler sometimes got stopped and not restarted.  In edge-preview, that can no longer happen because said handler never gets stopped, it just bails out early if there is nothing to do.  I suspect that the logic in question is independent of Due vs. Quintic.

justinclift, 2020-02-07:
@ril3y Yeah, it's an unknown so far.  The Due's are much more widely deployed than quintics, and the quintics use the new code anyway. :wink:

Johnr24, 2020-02-09:
So I tried the feritte cores, and I think that hasnt helped however if I enable smoothing and lower the torrence it appears in my tool-path settings it doesn't stop.