synthetos/g2 #452: DUE PWM to VFD Support

DynamicDumbA55, 2020-02-08:
I'm not sure if this has been covered anywhere; but I have set up a VFD configured to be externally controlled. I'm using UGS to control the spindle_PWM pin on the DUE. This is going into a circuit that converts PWM to 0-10v analog signal and then into the VFD. 
I seem to be unable to control the VFD's frequency/RPM. Not sure if this is a bug or perhaps I'm using the wrong commands? Am I able to change the duty cycle of the pin too?

justinclift, 2020-02-09:
Interesting.  This is something I want to do as well, for more powerful spindle motors than the "stock" ones I've been using so far.

Hopefully one of the other developers or a Community member has already gone through this process and gotten it working. :smile:

karoria, 2020-02-09:
Have you tested analog voltage output after your pwm to analog converter. It should change according to live rpm set in ugs. Also, make sure that your "run" setting is also configured to command externally and wired accordingly and you give signal for that through a relay (it should be isolated when in idle mode and conducted when in run mode). 
Most of the people have issues with getting in run mode. Once run mode is activated then it moves the spindle according to analog voltage input.

DynamicDumbA55, 2020-02-09:
I just re-wired everything and took some time to make sure what I was doing was correct. I relaunched UGS and tested. 
[image]
Here is the outputs in console when I run spindle reset override, spindle++ override, and a s### via the state tab.
I tested the PWM pin and analog pin and saw no changes in the magnitude, frequency, or duty cycle as I did all of these commands. 
Is there additional configuration that needs to be done in UGS? I could not find any information on it.

DynamicDumbA55, 2020-02-09:
Update: Using the {"p1"...} command, I can change pin 11's pwm and I can now see the frequency and rpm change. There was some extra set up required on the VFD which I got sorted last night and will document/make a video about for those seeking to do a similar set up. 
The m3 s### command seems to work; BUT, I'm not seeing the desired control that I would like. m5 results in a duty cycle of 10% which yields around 2500 RPM, m3 s1000 to 20000 seems to yield 3000-5000 RPM. I am fairly certain that my VFD is set properly as per guides and manufacturers documentation for external control and the wiring is correct.

After a decent nights rest, I've suddenly realized what 'p1' is.
{"p1":{"frq":100,"csl":7900,"csh":12800,"cpl":0.13,"cph":0.17,"wsl":7900,"wsh":12800,"wpl":0.13,"wph":0.17,"pof":0.1}}

P1 is used to set the PWM pin 11's defaults ('duh'). With pof being the default duty cycle. I'm not sure what csl/h, cpl/h, wsl/h, and wpl/h is and would love some clarification on what these mean. I have reason to believe that they are set points for RPM and duty cycle but am very unsure and do not want to start changing things without knowing what they are. 
Please let me know if you have any insight! Thank you so much!

karoria, 2020-02-09:
Ok. It seems g2core configuration issue. I have not touched g2core for almost a year now, but i can remember one of the branches at that time was giving default lowest spindle pwm duty cycle as 0.13 and highest was 0.17 making it difficult to notice any change in spindle rpm. Then i found that setting somewhere and changed it to 0 and 1 respectively and then it worked like a charm.

justinclift, 2020-02-09:
Sounds like progress is being made, and the full solution is *nearly* there...

karoria, 2020-02-09:
OMG, it is still present.. check above cpl, cph, wpl and wph. Change it to 0 (for low) and 1 (for high) respectively and set your spindle rpm as per spindle's specs. And I am sure, you will get it working.

karoria, 2020-02-09:
> OMG, it is still present.. check above cpl, cph, wpl and wph. Change it to 0 (for low) and 1 (for high) respectively and set your spindle rpm as per spindle's specs. And I am sure, you will get it working.

Sorry, I didn't read your previous post fully. You were already near to this...

DynamicDumbA55, 2020-02-09:
got the spindle up and running after changing those cpl/h... variables. However, on power off and restart, these values are wiped. Is there a way to flash the desired values to the board to not have to deal with rescaling every time??

justinclift, 2020-02-09:
> Or is there a way to initially run a set of commands on start up/ connection with the DUE?

Not with the Due board.  Instead, you'll need to compile the g2core code with the updated setting values (generally changed in ['settings_default.h'](https://github.com/synthetos/g2/blob/c684edcd38a18e4014860a9bf1ace1dc3b37285f/g2core/settings/settings_default.h)), then flash the Due with that.

There are step by step instructions for compiling and flashing, if that's useful?

&nbsp; &nbsp; https://github.com/synthetos/g2/wiki/Getting-Started-with-g2core-Development#compiling-the-code

DynamicDumbA55, 2020-02-11:
Update: 
I got the Atmel project and re-configured the settings_default.h as @justinclift mentioned and the UGS output is shown below.
[image]

However, I have one more question. As shown in the image below, "fbc" is set to settings_shapeoko2.h. 
[image]
What is this doing? where do I configure this in atmel? It doesn't seem to be affecting anything but I'd like confirmation of what the heck it is if possible.
Thank you all so much for your help! @justinclift @karoria

justinclift, 2020-02-11:
Interesting.  `fbc` shows the firmware configuration used (eg the name of the settings file).

Personally I'm used to compiling under Linux, by running this as the command line:

```
$ make PLATFORM=DUE BOARD=gShield
```

That doesn't use the Atmel IDE though.  On (say) Windows + the Atmel IDE, I'm not personally sure how to do it.  I think I tried it out once a few years ago, but have no real recollection of the steps. :man_shrugging:

If your newly compiled firmware is picking up the changes you made though (eg to `settings_default.h`), then I'd call this "good enough" and not be overly worried. :wink:

If the changes aren't being seen though, then updating `settings_shapeoko2.h` instead of `settings_default.h` would probably be the right approach. :smile:

giseburt, 2020-02-11:
I wouldn't recommend changing settings_default.h - those are literally the defaults, and anything changed there and set in the used settings file will be overridden, which could get very confusing. In this case, as indicated by fbc, the file settings_shapeoko2.h is being used. That is decided by the boards.mk file (for the CONFIG chosen on the make command line) or overridden on the make command line as SETTINGS_FILE.

In this case the easiest thing is to change settings_shapeoko2.h. If there's a setting in settings_default.h that you need to change, copy it to settings_shapeoko2.h and change it there. As for order of settings in the file, it mostly doesn't matter, as long as they don't refer to each other. If they do, what's being referred to has to be defined before it's being used. Otherwise, order isn't important, just whatever makes the most sense for readability - and if you're not publishing it, that's whatever the heck  you want :-).

-Rob

closed 2020-02-17 by DynamicDumbA55

justinclift, 2020-02-18:
@DynamicDumbA55 How'd you go with this? :smile:

DynamicDumbA55, 2020-02-21:
@justinclift So, I was able to get the PWM set by changing the values in settings_default.h.  However, atmel studios somehow decided to use settings_shapeoko2.h. So, I think I have some more poking around to do to see how to change this.... 
And by looking around these preset configs, I've figured out how to set up a fourth motor as a mirror of the Y axis (wooohoo). 

So in short, atmel confuses the heck out of me; but it seems to be working :)

And sorry for the delay to reply, been a but hectic on my end!

justinclift, 2020-02-21:
No worries about the delay, I'm several tasks behind at the moment myself. :wink:

And yeah, the main thing is you got it working, so all good. :grin:

Mgilbride, 2021-03-18:
So, a question here. Did you just update the spindle speed max to 24000? I am not seeing anything electric to update there.

DeanCavaliere, 2021-03-18:
Oh it's been an eternity since I looked at this project. I briefly scrolled through the settings file and for our application, we changed quite a few variables. Spindle max was one of them, but we also had to configure jerk, velocity, feedrate [maybe don't alter these until you decide your machine is moving too fast/slow for your liking], motors 1-4 as per the specs of the motors, and for PWM input we changed the P1_PWM settings [~ line 1050 - 1079] to match the specs of the VFD. 

Hopefully that answers your question?

Mgilbride, 2021-03-18:
Thanks. Yes. I did not notice the PWM settings down towards the bottom and was still looking for spindle max min settings as in older settings files. Much appreciated.