synthetos/g2 #454: where are the mathematics equations coming from?

SpaceView, 2020-02-17:
Can anyone give me some clue,
I just started to look into the g2core code, not familiar with it yet. 
In plan_zoid.cpp I found the following functions related to jerk calculation, e.g.
mp_get_target_length,
mp_get_target_velocity
mp_get_decel_velocity
_get_meet_velocity,
and, I checked lots of papers related to jerk control, and I cannot get the exact formula as the code provided, 
can anybody give me some clue or paper reference, where are those functions coming from?

justinclift, 2020-02-17:
Sounds like a question for @giseburt. :smile:

giseburt, 2020-02-17:
Not many people have dug that deep. :) There is no directly citable source, as most of those functions are independently derived for our own purposes.

The quintic velocity equations the we derived from are from the book [Pythagorean-Hodograph Curves: Algebra and Geometry Inseparable (Geometry and Computing)](https://www.amazon.com/dp/3540733973/ref=cm_sw_em_r_mt_dp_U_sqRsEbV8V5BAD) by  Rida T Farouki.

closed 2020-02-17 by giseburt