synthetos/g2 #455: motor control board for 4 motors (What are you guys using?)

DynamicDumbA55, 2020-02-17:
I've been looking into finding a compatible motor controller board that can be used to power and control 4 NEMA 23 stepper motors. 2 of these will be moving the X axis.
I already have a [gshield v5](https://synthetos.myshopify.com/products/gshield-v5); however I failed to realize that this board can only power and control 3 motors. I'm not entirely sure if it is possible, but do you think I would be able to get a 1-axis motor control board like [this from sparkfun](https://www.sparkfun.com/products/12859?_ga=2.116567494.1129919016.1513789857-1738791622.1506469362)?
I'm posting this to see what everyone else may be using to control their motors because it seems there are A LOT of options out there that could be fully compatible with g2core. I am on a budget and would love suggestions that are cheap but efficient (P.S. I'm running 24V to the NEMA 23's)
Thank you so much!

MitchBradley, 2020-02-17:
My 4-axis mill setup has a gShield running Z and A, which are not speed critical.  On X and Y, where I need more power, I use integrated closed loop steppers like this https://www.amazon.com/UCONTRO-iHSS57-36-20-Closed-Loop-Integrated-direction/dp/B074DTLDXC , and for the budget minded there is this https://www.amazon.com/Semoic-Printer-SERVO57B-Directly-Mainboard/dp/B081RFXF4X .  The more expensive one is, unsurprisingly, about twice as powerful as the cheaper one.

If you don't want to go the closed loop route you could use something like this https://www.amazon.com/STEPPERONLINE-1-0-4-2A-20-50VDC-Micro-step-Resolutions/dp/B06Y5VPSFN .

Either way, you would need to wire from pins on the microprocessor to pins on the driver.  I have had pretty good success directly driving my integrated steppers from Arduino Due GPIOs without any extra buffering.

MitchBradley, 2020-02-17:
Pololu has a lot of good stepper drivers too - https://www.pololu.com/category/120/stepper-motor-drivers .  For a single-motor project, I used this one with good success: https://www.pololu.com/product/2973 .  There is a smaller footprint version with that same chip at https://www.pololu.com/product/3096, but the larger one has on-board capacitors which is nice when you don't already have caps on the board it plugs into.  Don't be confused by peak current vs continuous current.  The sparkfun board says 2A but that is peak.  A4988 is about 1.2A continuous.  The TB67S249-FTG chip on the board I mentioned can do 1.6A continuous, 4.5A peak.

Johnr24, 2020-03-25:
> My 4-axis mill setup has a gShield running Z and A, which are not speed critical. On X and Y, where I need more power, I use integrated closed loop steppers like this [...]

Hi Mitch

How are you connecting these?
Can I see some pictures or a wiring diagram, 
I'm struggling to get mine to move despite hooking it up to the step direction pins

MitchBradley, 2020-03-25:
It would be hard to see what is going on in a photo because it is all in a compact box but I can explain it.

The gShield does not overhang the 36 pin 2 row connector on the end of the Due, so I hand-wired a breakout board that plugs into that connector.  On the outside row of that connector, I used pins 47, 45, 43, 41, 39, 37, 35, 33 for motors 4 and 5:

```
pin_number kSocket4_StepPinNumber           = 47;
pin_number kSocket4_DirPinNumber            = 45;
pin_number kSocket4_EnablePinNumber         = 41; // 43:limit

in_number kSocket5_StepPinNumber           = 39;
pin_number kSocket5_DirPinNumber            = 37;
pin_number kSocket5_EnablePinNumber         = 33; // 35:limit
```

The output pins are directly wired to the optocoupled inputs of hybrid steppers - GPIO pin to the optocoupler plus side, optocoupler minus side to GND.  I used RJ45 jacks to pass through the case.  A good wiring trick is to use a "keystone style" RJ45 inline coupler - female to female - going through the case panel, then you can cut a network cable in half and use one plug on the inside to go to the controller, the other plug on the outside to go to your drivers.

All of the GPIOs that are used are "high current" pins capable of sourcing 15mA/pin. (See http://www.robgray.com/temp/Due-pinout-WEB.png) The "low current" pins like GPIOs 22, 52, and some others, would not work.  This technique would also run out of steam if I had to drive several more outputs, because the total current would then exceed the Due's rating for total current from all I/O pins.  The limit would be about 10 pins directly driving optocouplers.  If I had to drive more, I would use something like ULN2803 to buffer between the Due and the optos.  I would also do that if my drivers happened not to work with 3.3V drive.

Johnr24, 2020-03-31:
Thanks
How are you managing the alarm circuit?
I was thinking it would be very clever/cool to have the alarm circuit trigger a subroutine that would

1. Pause the cut in the G-code
2. Re-home the system doing the Z-first followed by XY/other axis
3. Reposition the tool above the work piece in the correct area 
4. Resume the G-code and cut, I would imagine this would be automatic not sure how someone would do this as I'm not a coder