synthetos/g2 #456: Question: motor configuration

UweSauter, 2020-02-18:
I got a question regarding motor configuration when building the g2 firmware for a custom board.

In my settings file I have (taking motor 1 as example):
```
//*****************************************************************************
//*** Motor Settings **********************************************************
//*****************************************************************************

// MOTOR 1
#define M1_MOTOR_MAP                AXIS_X_EXTERNAL         // {1ma: AXIS_X, AXIS_Y...
#define M1_STEP_ANGLE               0.036                   // {1sa: degrees per step
#define M1_TRAVEL_PER_REV           5.0                     // {1tr:  1.25 is a typical value for a screw axis
#define M1_MICROSTEPS               1                       // {1mi:  1,2,4,8,    16,32 (G2 ONLY)
#define M1_STEPS_PER_UNIT           2000                    // {1su:  steps to issue per unit of length or degrees of rotation
#define M1_POLARITY                 0                       // {1po:  0=normal direction, 1=inverted direction
#define M1_ENABLE_POLARITY          IO_ACTIVE_LOW           // {1ep:  IO_ACTIVE_LOW or IO_ACTIVE_HIGH
#define M1_STEP_POLARITY            IO_ACTIVE_HIGH          // {1ps:  IO_ACTIVE_LOW or IO_ACTIVE_HIGH
#define M1_POWER_MODE               MOTOR_POWERED_IN_CYCLE  // {1pm:  MOTOR_DISABLED, MOTOR_ALWAYS_POWERED, MOTOR_POWERED_IN_CYCLE, MOTOR_POWERED_ONLY_
#define M1_POWER_LEVEL              0.0                     // {1pl:   0.0=no power, 1.0=max power
```
which can be queried on the running system:
```
g2core[mm] ok> 
$m
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               0.036 deg
[1tr] m1 travel per revolution    0.0000 mm
[1su] m1 steps per unit               inf steps per mm
[1mi] m1 microsteps               1 [1,2,4,8,16,32]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[1pl] m1 motor power level        0.000 [0.000=minimum, 1.000=maximum]
[1ep] m1 enable polarity          0 [0=active HIGH,1=active LOW]
[1sp] m1 step polarity            1 [0=active HIGH,1=active LOW]
```

As you can see `travel per revolution` and `steps per unit` is not set correctly.

It is clear to me that `steps per unit` is calculate by the software if `step angle` and `travel per revolution` are given. What is unclear is the behavior I should expect if `steps per unit` is also configured and which values I should give `step angle` and `travel per revolution` when I wish to define `steps per unit` myself (applying both to runtime configuration and settings file).

I'm using servo motors (JMC iHSV57-30-18-36-21-38) which can be configured to have between `800` and `40000` steps / rev and ball screws with 5mm / rev thread. With `10000 steps / rev` I was aiming for a positional accuracy of better than 0.001mm.

justinclift, 2020-02-20:
This is a bit confusing to me. :frowning:

In my configuration stuff (so far), I've always taken the "step angle" from the physical specs of the motor being used.  eg `1.8`  for the motor here:

&nbsp; &nbsp; https://www.omc-stepperonline.com/nema-17-stepper-motor/nema-17-bipolar-45ncm-63-74oz-in-1-5a-42x42x39mm-4-wires-w-1m-cable-and-connector.html

Then I define the "travel per rev" setting, to match the physical distance moved when turning the motor.

He's an example of a (working) config for a machine in my local Makerspace:

```
// MOTOR 1
#define M1_MOTOR_MAP                AXIS_X
#define M1_STEP_ANGLE               1.8
#define M1_TRAVEL_PER_REV           4.0
#define M1_MICROSTEPS               8
#define M1_POLARITY                 0                       
#define M1_POWER_MODE               MOTOR_POWERED_IN_CYCLE
#define M1_POWER_LEVEL              0.5
```

Does that help? :smile:

karoria, 2020-02-20:
I think, steps per unit is derived value. So either enter steps per unit directly or enter all other fields will do the job.

UweSauter, 2020-02-20:
I tried commenting:
```
// MOTOR 1
#define M1_MOTOR_MAP                AXIS_X_EXTERNAL         // {1ma: AXIS_X, AXIS_Y...
//#define M1_STEP_ANGLE               0.036                   // {1sa: degrees per step
//#define M1_TRAVEL_PER_REV           5.0                     // {1tr:  1.25 is a typical value for a screw axis
#define M1_MICROSTEPS               1                       // {1mi:  1,2,4,8,    16,32 (G2 ONLY)
#define M1_STEPS_PER_UNIT           2000                    // {1su:  steps to issue per unit of length or degrees of rotation
#define M1_POLARITY                 0                       // {1po:  0=normal direction, 1=inverted direction
#define M1_ENABLE_POLARITY          IO_ACTIVE_LOW           // {1ep:  IO_ACTIVE_LOW or IO_ACTIVE_HIGH
#define M1_STEP_POLARITY            IO_ACTIVE_HIGH          // {1ps:  IO_ACTIVE_LOW or IO_ACTIVE_HIGH
#define M1_POWER_MODE               MOTOR_POWERED_IN_CYCLE  // {1pm:  MOTOR_DISABLED, MOTOR_ALWAYS_POWERED, MOTOR_POWERED_IN_CYCLE, MOTOR_POWERED_ONLY_WHEN_MOVING
#define M1_POWER_LEVEL              0.0                     // {1pl:   0.0=no power, 1.0=max power
```
but it doesn't take the supplied `steps per unit`:
```
g2core[mm] ok> 
$m
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               1.800 deg
[1tr] m1 travel per revolution    0.0000 mm
[1su] m1 steps per unit               inf steps per mm
[1mi] m1 microsteps               1 [1,2,4,8,16,32]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[1pl] m1 motor power level        0.000 [0.000=minimum, 1.000=maximum]
[1ep] m1 enable polarity          0 [0=active HIGH,1=active LOW]
[1sp] m1 step polarity            1 [0=active HIGH,1=active LOW]
```
It even used some other defaults to fill `step angle`.

If I comment `steps per unit` and fill `step angle` and `travel per revolution` I get:
```
$m
[1ma] m1 map to axis              0 [0=X,1=Y,2=Z...]
[1sa] m1 step angle               0.036 deg
[1tr] m1 travel per revolution    5.0000 mm
[1su] m1 steps per unit        2000.00012 steps per mm
[1mi] m1 microsteps               1 [1,2,4,8,16,32]
[1po] m1 polarity                 0 [0=normal,1=reverse]
[1pm] m1 power management         2 [0=disabled,1=always on,2=in cycle,3=when moving]
[1pl] m1 motor power level        0.000 [0.000=minimum, 1.000=maximum]
[1ep] m1 enable polarity          0 [0=active HIGH,1=active LOW]
[1sp] m1 step polarity            1 [0=active HIGH,1=active LOW]
```
I don't understand why there is `#define M1_STEPS_PER_UNIT` if I can't use it to enter the value directly.

In case of stepper motors, it probably is easier to let the system(?)/compiler(?) calculate `steps per unit` but with the servo motors where I can adjust the steps per revolution and where I have no microstep setting, I'd find it easier to just enter `steps per unit` directly.

giseburt, 2020-02-20:
Steps per unit was added by ShopBot, and they only ever used it by setting `su` at runtime. I don’t believe the current code is setup to use it properly at initialization time.

That said, `su` really should be avoided, as it bypasses several layers of sanity checks that we have in place.

If you omit `M1_STEPS_PER_UNIT` and leave just this:
```
#define M1_STEP_ANGLE               0.036                   // {1sa: degrees per step
#define M1_TRAVEL_PER_REV           5.0                     // {1tr:  1.25 is a typical value for a screw axis
#define M1_MICROSTEPS               1                       // {1mi:  1,2,4,8,    16,32 (G2 ONLY)
```

You can verify that it works as expected by querying the motor and seeing that you get 2000 as steps/unit.

As not, and this is likely less to you and more to others reading, `STEP_ANGLE` is for whole steps. In this case, the `MICROSTEPS` is set to `1`, but if it were any other value the steps/unit would be adjusted assuming that `STEP_ANGLE` actually represented `MICROSTEPS` number of “steps” (according to pulsing the step pin).

UweSauter, 2020-02-20:
Thanks for that clarification. Might be worth a comment in the settings files.

closed 2020-02-20 by UweSauter