synthetos/g2 #457: how to use this with 5V stepper drivers?

crazyxangel, 2020-03-07:
i am finding 20 diffirent answers to this problem online all saying something diffirent from the other
so i thought i'd ask it here
Arduino due with g2core how do you guys interface it with 5V logic level stepper drivers?

MitchBradley, 2020-03-07:
Since you directed the question to "you guys", don't be surprised if you get 20 different answers, one from each "guy".  The answer depends on which particular stepper driver you are using.  In my case I am using integrated closed loop steppers like IHSV57-30-10 .  They have optocoupler inputs that nominally need 5V to drive, but I have found that they work just fine when driven directly from 3V3 Due GPIOs.  The optocoupler drive current at 3v3 compared to 5V is less than ideal, but nevertheless it is enough for reliable switching.

If you are using Pololu-stick driver modules, most of them are specified to work just fine with either 3v3 or 5v logic drive.

Ask a vague question, get a vague answer.  There are many different "5V logic level stepper drivers".

MitchBradley, 2020-03-07:
If you have a driver that really, really, needs 5V drive, then you need a level translator.  The type of level translator depends on the type of input the driver has.  If it is optocoupled and 3V3 won't switch it, then you can add one or more 8-channel ULN2803A chips to drive the optocoupler low sides, feeding the high sides from 5V.  Or you can use individual MOSFETs or bipolar transistors, one per signal.  Or you can drive the optocoupler high sides with any buffer chip that can accept 3V3 logic inputs and drive to 5V on the output, for example 74LVC245.  There is an endless number of workable ways to do this.

MitchBradley, 2020-03-08:
#443  has an extensive discussion of different approaches to driving steppers from a Due, with schematics and discussion of pros and cons. I'm closing this one as a duplicate of the earlier one.

closed 2020-03-08 by MitchBradley

justinclift, 2020-03-08:
@crazyxangel If you're ok with spending a bit of cash on a board specifically for this purpose, then these are apparently good:

https://webshop.djuke.nl/kits/g2shield-external-kit

That's the board for external drivers, eg the ones that have their own enclosure, like this:

&nbsp; &nbsp; http://leadshine.com/ProductSubType.aspx?type=products&category=stepper-products&producttype=stepper-drives&subtype=general-stepper-drives