synthetos/g2 #458: Question on feedrate config values

DeanCavaliere, 2020-03-13:
Depending on which settings_<machine name> you look at on here, you can see different values of things like jerk, feedrate, velocity, etc etc. The values of x and y are typically exactly the same but z is usually different. For example, in settings_shapeoko2dual, the x axis has a velocity/feedrate of 50000 while the z axis has a value of 1200. 
Why is this? is there a relationship between these values so that the x/y and z axis all move at the same relative speeds in reality? or can these values be arbitrarily chosen?

[image]

justinclift, 2020-03-13:
It's a "maximum" feed rate, and - for me at least - I tend to set it to what the machine seems happy with on that particular axis.

eg see how fast any particular axis seems to move in a reliable fashion, + repeatable, then set the feed rate maximum for that axis to that value

The difference of X & Y axis feedrates compared to Z axis is probably due to different tech being used on the example settings.  For example, it's pretty common to have belt driven motors run the X and Y axis, with the Z axis being ball screw driven.  Belt driven setups seem to have much higher potential feed rates than ball screws.  So, there's generally a difference there.

Does that help? :smile:

DeanCavaliere, 2020-03-13:
Yes that helps so much; I got the impression there might've been a relationship due to the different modes of travel being used but if it's just dependent on machine performance then perfect. Thank you so much man

closed 2020-03-13 by DeanCavaliere

justinclift, 2020-03-13:
You're welcome @DeanCavaliere. :smile: