synthetos/g2 #462: Feedrate not working

comm64, 2020-04-11:
Hello,
I was able to configure and have my g2core cnc running very well. I have left only one little issue and I would like to understand if I am doing something wrong.
I am using CNCJS with flow control enabled.
In my setting i have velocity set to 2000
and feedrate set to 2000 to.
Now when i run a gcode or if i manually send a comnmand like G0 X200 F250 trhe machine moves always at the speed set in velocity that in this case is 2000.
What I am missing here and what I am doing wrong?
Thank you!

ril3y, 2020-04-11:
G0 does not take feed rates.  It is a traverse.  It will move at the velocity max setting.

Do a G1 X200 250 and you will see it act as you are expecting.

closed 2020-04-11 by ril3y

justinclift, 2020-04-11:
@comm64 How'd you go with that? :smile:

comm64, 2020-04-11:
because I am an idiot lol! I think staying recluse in the house because this coronavirus situation is driving me nuts LOL.
Anyway thank you now my machine is working like a charm! I even got to work the probing. time to cut some stuff to kill the time

justinclift, 2020-04-12:
Heh Heh Heh. :smile:

Sounds like a good plan. :wink:

ril3y, 2020-04-12:
Happens. Don't worry.  Would love to see what you are using G2 on!

comm64, 2020-04-12:
I’m using the g2core with external dm542 drivers on a simple Chinese 3040 cnc . I learned a lot using this machine mostly because I always needed to make improvements to it . I’m using now this machine to make a better one