synthetos/g2 #463: Persistence of settings

AnthonyMyatt, 2020-04-13:
Hello,

Should settings for the motors/axis be persisting through a power cycle or is it normal to have to set these every time?

justinclift, 2020-04-13:
Depends on your hardware. :smile:

If you're using a Due, the settings don't persist across power cycles as the Due doesn't have any onboard flash (or other storage) for saving them.  The general approach there is to experiment and find out the settings you need, then recompile the firmware (not too hard on Linux or macOS) with those values as the defaults.

Does that help?

AnthonyMyatt, 2020-04-13:
Thanks @justinclift, that explains it.

justinclift, 2020-04-13:
No worries at all.  Hope the PrintNC build goes well. :smile:

closed 2020-04-13 by justinclift

AnthonyMyatt, 2020-04-13:
haha. Thanks. 😀

justinclift, 2020-04-14:
@AnthonyMyatt As a general thought, is the server at http://www.anthonymyatt.net supposed to be online?

Tried it last night, and again just now, and am not able to connect to it.  Seems like server problem of some sort. :wink:

AnthonyMyatt, 2020-04-14:
@justinclift, I once had a website running there with a bit of a bio and some of my work however I pulled it down some time ago as my working life changed. Now I only use the domain for email. I might reinstate it though if my new hobbies take off.

justinclift, 2020-04-14:
No worries.  Mostly asking because it seems like you've got a blog or something where you're documenting the build:

&nbsp; &nbsp; https://github.com/AnthonyMyatt/ImperativeMyatt-PrintNCBuild

Wondering what the URL is, as it'd be easier than my just looking through the photo's one at a time.  That's just me being lazy though. :grin:

AnthonyMyatt, 2020-04-14:
[https://anthonymyatt.github.io/ImperativeMyatt-PrintNCBuild/](https://anthonymyatt.github.io/ImperativeMyatt-PrintNCBuild/)

justinclift, 2020-04-14:
Awesome, thanks. :smile:

justinclift, 2020-04-14:
Oh cool, there's a cost breakdown too.  Super awesome, as it has Aus places to get stuff from (I'm in Melb too atm).

The PrintNC looks very interesting, as I've been thinking to use steel box section for a potentially cheaper CNC frame approach for a while.  And they've already gone through the process of getting that working.

justinclift, 2020-04-14:
With the Due, what sort of motor drivers are you planning to use?

If you're going for external drivers (eg [DM542](https://www.omc-stepperonline.com/digital-stepper-driver-10-42a-20-50vdc-for-nema-17-23-24-stepper-motor-dm542t.html) or similar), you'll very, very likely want one of these to interface between the Due and the drivers:

https://webshop.djuke.nl/kits/g2shield-external-kit

AnthonyMyatt, 2020-04-14:
Looking to use TB6600 drivers. I ordered the g2Shield from djuke on the weekend.  😀

justinclift, 2020-04-14:
Awesome. :smile: