synthetos/g2 #464: 4th Axis Post Processor for Fusion 360

Johnr24, 2020-04-23:
Hi 
does anyone have a 4th axis (rotary) post for fusion 360?
Best
John

justinclift, 2020-04-24:
Not so far.  It's something I've looked into before as well, by modifying the existing 3-axis post.  I wasn't able to get the retraction (Z axis) height working properly when I tried though.

Someone else may have already done it though, and figured it out.

Johnr24, 2020-04-24:
Hey 

> Not so far. It's something I've looked into before as well, by modifying the existing 3-axis post. I wasn't able to get the retraction (Z axis) height working properly when I tried though.
> 
> Someone else may have already done it though, and figured it out.

I did some digging and found one on the Fusion 360 Forum

Google search "Tiny G 4th.cps"

I've just done a test with a 3D adaptive toolpath and a rotary toolpath
I was able to do get both done, I did have to set my clearance height to stock top, in order to make it work, 

[image]

justinclift, 2020-04-24:
Interesting.  It's not coming up for me.  Would you be ok to share the link to the Fusion 360 forum post?

justinclift, 2020-04-24:
Ahhh, seems like this one:  https://forums.autodesk.com/t5/fusion-360-manufacture/simple-4-axis-setup-and-post-processor-failure/m-p/6694718#M20336

I'll give that a go in a few weeks/months, after the local Makerspace opens up again.

Thanks heaps @Johnr24. :smile:

Johnr24, 2020-04-24:
I absolutely would have shared it but I found it last night!
That is the one, and actually it's the one lower down on that page as it's been tweaked, Now I need to decide on which 4th axis to get!

Johnr24, 2020-08-10:
I've since added tweaked this processor myself and added a pull request to get it to be part of G2cores resources folder 
https://github.com/synthetos/g2/pull/477

Best
J