synthetos/g2 #466: gShield pinout doesn't appear to match gShield

bobobo1618, 2020-05-14:
[`gShield-pinout.h`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/gShield-pinout.h) references 66 different pins. The gShield [has 28](https://github.com/synthetos/grblShield).

I thought maybe this might be another shield with the same name but [`ArduinoDue.mk`](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue.mk#L27-L28) specifically says it's for the Synthetos gShield, which is the one above and the one I bought.

What's going on?

ril3y, 2020-05-14:
The first .h file is for the Arduino-Due not the gShield.  The 2nd is the gShield.

closed 2020-05-14 by ril3y

bobobo1618, 2020-05-14:
What do "first .h file" and "2nd" refer to?

Do you mean `gShield-pinout.h` refers to something other than the gShield?

ril3y, 2020-05-14:
Yes you listed a .h file and then you gave a link.  The gShield-pinout.h is
a setup for the Arduino-Due to use the gShield.  The 2nd link is a link to
the pins on the gShield.

bobobo1618, 2020-05-14:
Yes, that was deliberate. What I'm trying to point out is that the file that sets up the Arduino Due to use the gShield refers to a much larger number of pins than the gShield itself has.

This is confusing because the gShield should have everything one needs to interact with one's machine. It's unclear what the extra pins are for, what they do and how they're intended to be used.