synthetos/g2 #467: SPI API looks like overkill for simple startup config

bobobo1618, 2020-05-14:
I've been trying to get the TMC5160 driver running with a nice clean starting point but Motate's API appears to be async-only.

I understand that this makes a lot of sense for certain applications but for a simple set of oneoff startup commands it seems like a lot of complexity without much benefit.

Would there be any issues with adding a simpler synchronous API?

giseburt, 2020-05-14:
Having a synchronous API means maintaining and supporting both, and the only time g2core would be able to use a synchronous API is during boot or in a time where it’s guaranteed to not have motion. Boot time is fine, but we try to avoid and discourage anything written requiring lack of motion.

I presume you’re basing the 5160 driver on https://github.com/synthetos/g2/blob/edge/g2core/device/trinamic/tmc2130.h and .cpp - the basic framework should already be handled. I understand that it’s more complex than a synchronous style but it’s worth it to get the responsiveness.

I have been watching C++20 coroutines that will help simplify this type of asynchronous programming a fair bit. I don’t think we’re quite there yet though.

bobobo1618, 2020-05-15:
That sounds reasonable enough, it's just really frustrating. I'm at the point where I'd like to "just" get a proof of concept going for this driver and this is just one of the many obstacles I've run into so far.

Setting up a basic demo of the driver with the Arduino IDE was trivial.

closed 2020-05-15 by bobobo1618