synthetos/g2 #468: How can I compile firmware for a V9 board board153v9k?

bartona, 2020-05-28:
I have a Synthetos TinyG v9 board153v9k PCB, and I would like to put the latest g2core firmware on.  Is this board supported as a target?  Do I need to use the g2 firmware, or the original TinyG firmware for this board?

justinclift, 2020-05-28:
From rough memory, as I've not used the v9 TinyG board myself, it does use g2core for it's firmware.  Found a direct mention of things here, which confirms it:

&nbsp; &nbsp; https://synthetos.com/topics/tinyg-v9

With your board, is it part of a known machine (eg a ShopBot of some model, etc)?

Asking because looking through the `boards.mk` file for potential compile targets, I'm not sure which one is close to the settings you'd be needing:

&nbsp; &nbsp; https://github.com/synthetos/g2/blob/c684edcd38a18e4014860a9bf1ace1dc3b37285f/g2core/boards.mk

Note the machines there with an entry of `BOARD=g2v9k`.  I **think** that "g2v9k" is your board.  So, one of those machines is probably along the right lines. :wink:

giseburt, 2020-05-28:
@justinclift is correct. `g2v9k` is the v9 you have.

If I may: how did you come to have a v9 board? I assume it’s in something. 😄

scottnaylor, 2020-05-31:
Hello -
I am in a similar situation - I have a g2 V9 board in my Printrbot Crawlbot. 
The original fw info below (unfortunately I can't find the original bin anymore either) 
[fb]  firmware build             89.03
[fbs] firmware build "                           build"
[fv]  firmware version            0.98
[cv]  configuration version       7.00
[hp]  hardware platform           3.00
[hv]  hardware version            0.00
[id]  TinyG ID       0213-0215-d413