synthetos/g2 #469: Configure Spindle PWM Control on ArduinoDue gShield

lumascet, 2020-05-28:
I have troubles configuring the correct pins for PWM spindle speed control on the ArduinoDue on edge-preview branch. My hardware follows the ArduinoDue Pinout: [https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core](url). 

so i added these lines to my settings file:
```
#define SPINDLE_ENABLE_OUTPUT_NUMBER 	12
#define SPINDLE_DIRECTION_OUTPUT_NUMBER 13
#define SPINDLE_PWM_NUMBER 		11
#define MIST_ENABLE_OUTPUT_NUMBER 	57
```

Building and uploading results in not being able to connect to the board via uart. am i missing something? Hardware errors can be ruled out, since it had worked in the edge branch.

justinclift, 2020-05-29:
Which model of PWM are you using? :smile:

lumascet, 2020-05-29:
> 
> 
> Which model of PWM are you using? 😄

I'm not sure if I understand your question correctly :sweat_smile:, the pwm should be controlled via the p1 property i think, these are my settings of it:
```
{"r":{"p1":{"frq":10000,"csl":0,"csh":24000,"cpl":0,"cph":1,"wsl":0,"wsh":24000,"wpl":0,"wph":1,"pof":0}},"f":[1,0,8]}
```
I previously didn't have to alter the default gShield pinout configuration as it worked out of the box: [gShield-pinout.h](https://github.com/synthetos/g2/blob/edge/g2core/board/ArduinoDue/gShield-pinout.h#L211)
Im not sure what exactly changed that these settings are not working anymore in edge-preview.

justinclift, 2020-05-29:
Gah.  That's a brain-o on my part.  Was thinking VFD (like [these](https://www.omc-stepperonline.com/variable-frequency-drive/)) for some reason.  Doh. :facepalm:

lumascet, 2020-05-29:
> 
> 
> Gah. That's a brain-o on my part. Was thinking VFD (like [these](https://www.omc-stepperonline.com/variable-frequency-drive/)) for some reason. Doh. 🤦

Oh, yes indeed im using it in combination with a VFD like this one, with 0-10V is 0-24k rpm
The 0-3V3 to 0-10V conversion i realized through an Optocoupler.

justinclift, 2020-05-29:
Cool!

@giseburt has assigned this to himself, so you're in good hands when we has time to jump in.

I've not yet personally used a VFD at all, but I intend to when I get back into doing this stuff in a few months time. :grin:

lllars, 2020-06-30:
I've also run into this issue with Due on edge-preview.  I tried different settings:

```
#define SPINDLE_ENABLE_OUTPUT_NUMBER        4
#define SPINDLE_DIRECTION_OUTPUT_NUMBER     5
#define SPINDLE_PWM_NUMBER                  6
```

I'm not sure if output number is supposed to correspond to pin number.  I'm trying to use pin 11 on the Due.

lumascet, 2020-07-16:
> I've also run into this issue with Due on edge-preview. I tried different settings:
>
> ```
> #define SPINDLE_ENABLE_OUTPUT_NUMBER        4
> #define SPINDLE_DIRECTION_OUTPUT_NUMBER     5
> #define [...]

Have you figured out anything yet?