synthetos/g2 #470: incorrect input pin state after glitch

lllars, 2020-06-18:
I found a bug in the input pin debounce / lockout code.  If an input pin changes, and then changes back before the lockout time has expired, then it's state is never updated to reflect that it changed back.

As an example, on my machine, I have hall effect limit switches which are normally high.  I first power up g2core by connecting it to my PC.  Then, when I power on the machine, the switch state very briefly goes low, only for a few ms, and then returns to high.  g2core correctly detects that that the switches go low, but then never updates that they have gone back to high: $in continues to display the incorrect state.  Because of this g2core then fails to detect the next time the switches go low, but it does then update to the correct state when the switch goes back to high.  This bug causes my machine to crash when homing.

I've worked around this issue by changing pin_changed() in gpio.cpp.  I moved the code to record the pin state change up, so that the state change is recorded even if the pin is locked out.  The lockout check is still there and prevents any actions from occurring during lockout even though the state has changed.

I am on firmware version 0.99, build 101.3.  Target board is an Arduino Due with no shield.  This is a 3 axis mill with independent stepper drivers.

Here is my re-ordering of the relevant section of gpio.cpp:
```
    void pin_changed() {
        if (D_IN_CHANNELS < ext_pin_number) { return; }

        d_in_t *in = &d_in[ext_pin_number-1];

        // return if input is disabled (not supposed to happen)
        if (in->mode == IO_MODE_DISABLED) {
            in->state = INPUT_DISABLED;
            return;
        }

	// return if no change in state
        bool pin_value = (bool)input_pin;
        int8_t pin_value_corrected = (pin_value ^ ((int)in->mode ^ 1));    // correct for NO or NC mode
        if (in->state == (ioState)pin_value_corrected) {
            return;
        }

	// record the changed state (we know it changed or we would have exited beforehand)
        in->state = (ioState)pin_value_corrected;
        if (pin_value_corrected == INPUT_ACTIVE) {
            in->edge = INPUT_EDGE_LEADING;
        } else {
            in->edge = INPUT_EDGE_TRAILING;
        }

        // return if the input is in lockout period (take no action)
        if (in->lockout_timer.isSet() && !in->lockout_timer.isPast()) {
            return;
        }

        // lockout the pin for lockout_ms
        in->lockout_timer.set(in->lockout_ms);

        // perform homing operations if in homing mode
        ...
```

justinclift, 2020-06-19:
Interesting.  The latest development code (in the `edge-preview` branch) has some substantial changes in parts.  Not sure if any of the gpio code was touched though.  It might be worth seeing if that has the same problem or not. :smile:

@giseburt Thoughts?

giseburt, 2020-06-19:
Indeed, Justin, that was one reason for the refactor. That problem should be resolved in the edge-preview branch, along with other substantial improvements in GPIO.

-Rob

justinclift, 2020-06-19:
Awesome. :smile:

@lllars Any interest in compiling in the `edge-preview` branch, and seeing how that goes?  It should work fine with the Arduino Due. :smile:

mhlong10, 2020-06-19:
I hate to say I ran into this same problem a few months ago but didn't bother to address it.  I simply manually touched the touch plate to the z-probe if I saw that it was out of sync (which got it back in sync) and then probed again.

Looking at the change my concerns are that in->state is actually bouncing around following the actual state of the pin.  gpio_read_input() and io_get_input() asynchronously read in->state and thus could get a "random" value depending on when called.  I think this is probably benign for io_get_input, and it looks like gpio_read_input is only called when the machine is in a static state.  So for current uses this change could be okay.  However,  the probing, homing, and subsequent actions will still miss reacting to very short pin transitions and fail to take appropriate action.  This is probably no worse that what is happening today.

I do like the simplicity of the fix.  I had a more elaborate / complete fix in mind but didn't get to it.

The plan I had was to start a retriggerable asynchronous timer for lockout_timer period of time whenever the pin state changes and once it expires call pin_changed() again to get the final resting state.  This has the affect of stretching a pin transition for at least lockout_timer period of time, thus locking out other changes during that time.  I've used this technique many times to debounce gpio pins on other projects.  Unfortunately I'm not aware of any interrupt driven timer within the code so this wasn't straightforward to implement and I had to move on to my real but less fun job.

lllars, 2020-06-19:
I'm happy to test out edge-preview.  Is there a summary anywhere of what changes it incorporates?

@mhlong10 I was also thinking that checking the state of the pin after the lockout_timer expired would be a good solution.  I just couldn't figure out an easy way to do that.

lllars, 2020-06-21:
I'm getting a lot of errors when trying to compile edge-preview.  I'll post the output I'm seeing.  Let me know if there's anything I can try on my end to fix things.  I'm compiling with:

```
make PLATFORM=DUE BOARD=gShield SETTINGS_FILE="settings_dynamyte_4000.h"
```

and the output is:

```
"Found that we're in Linux"
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.c
    -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/exceptions.o 
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.c
    -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/system_sam3x.o 
Compiling c ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.c
    -> build/gShield/motate/cmsis/TARGET_Atmel/sam3x/source/templates/gcc/startup_sam3x.o 
Compiling c ../Motate/MotateProject/motate/platform/atmel_sam/cortex_handlers.c
    -> build/gShield/motate/platform/atmel_sam/cortex_handlers.o 
Compiling c ../Motate/MotateProject/motate/platform/atmel_sam/hooks.c
    -> build/gShield/motate/platform/atmel_sam/hooks.o 
Compiling cpp ../Motate/MotateProject/motate/MotateUtilities.cpp
    -> build/gShield/motate/MotateUtilities.o 
Compiling cpp ../Motate/MotateProject/motate/main.cpp
    -> build/gShield/motate/main.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamDMA.cpp
    -> build/gShield/motate/Atmel_sam_common/SamDMA.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamPins.cpp
    -> build/gShield/motate/Atmel_sam_common/SamPins.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamPower.cpp
    -> build/gShield/motate/Atmel_sam_common/SamPower.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamServiceCall.cpp
    -> build/gShield/motate/Atmel_sam_common/SamServiceCall.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamSPI.cpp
    -> build/gShield/motate/Atmel_sam_common/SamSPI.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamTimers.cpp
    -> build/gShield/motate/Atmel_sam_common/SamTimers.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamUART.cpp
    -> build/gShield/motate/Atmel_sam_common/SamUART.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam_common/SamUniqueID.cpp
    -> build/gShield/motate/Atmel_sam_common/SamUniqueID.o 
Compiling cpp ../Motate/MotateProject/motate/Atmel_sam3x/SamUSB.cpp
    -> build/gShield/motate/Atmel_sam3x/SamUSB.o 
Compiling cpp ../Motate/MotateProject/motate/platform/atmel_sam/syscalls.cpp
    -> build/gShield/motate/platform/atmel_sam/syscalls.o 
Compiling cpp ./alarm.cpp
    -> build/gShield/./alarm.o 
In file included from ./g2core.h:62:0,
                 from ./alarm.cpp:29:
./board/ArduinoDue/hardware.h:47:5: warning: "HAS_HOBBY_SERVO_MOTOR" is not defined [-Wundef]
 #if HAS_HOBBY_SERVO_MOTOR
     ^
In file included from ./board/ArduinoDue/hardware.h:32:0,
                 from ./g2core.h:62,
                 from ./alarm.cpp:29:
./config.h:341:38: error: declaration of 'operator[]' as non-function
     const cfgItem_t& operator[](std::size_t idx) const;
                                      ^
./config.h:341:31: error: expected ';' at end of member declaration
     const cfgItem_t& operator[](std::size_t idx) const;
                               ^
./config.h:341:45: error: expected ')' before 'idx'
     const cfgItem_t& operator[](std::size_t idx) const;
                                             ^
./config.h:349:46: error: 'get' declared as a 'virtual' field
     virtual const cfgItem_t * const get(std::size_t idx) const;
                                              ^
./config.h:349:37: error: expected ';' at end of member declaration
     virtual const cfgItem_t * const get(std::size_t idx) const;
                                     ^
./config.h:349:53: error: expected ')' before 'idx'
[... 107 lines cut ...]
In file included from ./safety_manager.h:36:0,
                 from ./alarm.cpp:33:
./gpio.h:1100:43: error: call of overloaded 'abs(float&)' is ambiguous
         return std::sqrt(std::abs(variance));
                                           ^
In file included from /home/lars/dynamyte/g2/Motate/Tools/linux/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/ctype.h:4:0,
                 from ./g2core.h:29,
                 from ./alarm.cpp:29:
/home/lars/dynamyte/g2/Motate/Tools/linux/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/stdlib.h:66:5: note: candidate: int abs(int)
 int _EXFUN(abs,(int));
     ^
In file included from /home/lars/dynamyte/g2/Motate/Tools/linux/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/c++/5.4.1/bits/stl_algo.h:59:0,
                 from /home/lars/dynamyte/g2/Motate/Tools/linux/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/c++/5.4.1/algorithm:62,
                 from ../Motate/MotateProject/motate/MotatePins.h:34,
                 from ./board/ArduinoDue/hardware.h:76,
                 from ./g2core.h:62,
                 from ./alarm.cpp:29:
/home/lars/dynamyte/g2/Motate/Tools/linux/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/c++/5.4.1/cstdlib:166:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^
/home/lars/dynamyte/g2/Motate/Tools/linux/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/c++/5.4.1/cstdlib:174:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^
make: *** [../Motate/MotateProject/motate/Motate.mk:471: build/gShield/./alarm.o] Error 1
```

justinclift, 2020-06-21:
It's **probably** caused by something in the `settings_dynamyte_4000.h` file. :angel:

When I hit stuff like that, I generally first try to find a settings file that compiles ok or start with the default one (`settings_default.h`, used as a fallback when no other settings file is specified).

So, try without giving any settings file on the command line, and see if that compiles successfully:

```
$ make PLATFORM=DUE BOARD=gShield 
```

That should compile ok, and use the above mentioned default settings. :smile:

**If** it does so, and the `settings_dynamyte_4000.h` file has the right pieces you actually want... the next step is figuring out what's wrong in the dynamyte settings file.  I'd do that by making it just a direct copy of the default settings one, and then making a few changes to it at a time to figure out what breaks.  It shouldn't take more than a few compile attempts to determine the culprit.

In theory. :smile:

lllars, 2020-06-21:
Thanks, good suggestion, but it didn't work.  Output was identical.

```
$ make PLATFORM=DUE BOARD=gShield SETTINGS_FILE="settings_dynamyte_4000.h" clean
"Found that we're in Linux"
rm -fR build/gShield
rm -fR bin/gShield
rm -fR gShield.elf gShield.map gShield.hex gShield.bin
rm -fR compile_commands.json ../Motate/MotateProject/motate/compile_commands.json
$ make PLATFORM=DUE BOARD=gShield
```

justinclift, 2020-06-21:
Yep, you're right.  Just tried it here, and same error.

@giseburt Looks like `edge-preview` has become busted.  Something about `HAS_HOBBY_SERVO_MOTOR` maybe?

giseburt, 2020-06-21:
Checking the build now.

Also re:

> Unfortunately I'm not aware of any interrupt driven timer within the code so this wasn't straightforward to implement and I had to move on to my real but less fun job.

The homing and probing are all interrupt driven now. The debounce code was causing so much trouble because sometimes a probe would only briefly fire a pulse and the subset backoff would be fast enough that the next time we checked it wasn't triggered anymore.

See here for the handler:  https://github.com/synthetos/g2/blob/d6ae0db69db2e5e6f02294a599f1318dd967817a/g2core/cycle_probing.cpp#L132-L147 <https://github.com/synthetos/g2/blob/d6ae0db69db2e5e6f02294a599f1318dd967817a/g2core/cycle_probing.cpp#L132-L147>
Here for the setup (registration for interrupt events): https://github.com/synthetos/g2/blob/d6ae0db69db2e5e6f02294a599f1318dd967817a/g2core/cycle_probing.cpp#L361-L362 <https://github.com/synthetos/g2/blob/d6ae0db69db2e5e6f02294a599f1318dd967817a/g2core/cycle_probing.cpp#L361-L362>
And here for the teardown (reregistration): https://github.com/synthetos/g2/blob/d6ae0db69db2e5e6f02294a599f1318dd967817a/g2core/cycle_probing.cpp#L295-L296 <https://github.com/synthetos/g2/blob/d6ae0db69db2e5e6f02294a599f1318dd967817a/g2core/cycle_probing.cpp#L295-L296>

The interrupt handling itself is in Motate, but in g2core it comes in here: https://github.com/synthetos/g2/blob/1e2d02c460e5f188f204edd455246c78eaf3e09e/g2core/gpio.h#L486-L525 <https://github.com/synthetos/g2/blob/1e2d02c460e5f188f204edd455246c78eaf3e09e/g2core/gpio.h#L486-L525>

There is a lockout timer, so multiple interrupts are not fired for the same pin in a brief period. Note that we're not glitch filtering, but instead ignoring anything after the initial event for a brief amount of time, trusting that the initial event was genuine.

If there are noise spikes on the lines causing glitches then you'll need to filter it in hardware. Luckily this is fairly simple.

-Rob

giseburt, 2020-06-21:
Ok, that should be fixed in the latest commit.

I tested with compiling for: make CONFIG=MiniMillgShield

Which is defined here: https://github.com/synthetos/g2/blob/404106b75e95a3a12c981d2e1ff42ad626c3e015/g2core/boards.mk#L75-L81 <https://github.com/synthetos/g2/blob/404106b75e95a3a12c981d2e1ff42ad626c3e015/g2core/boards.mk#L75-L81>

-Rob

lllars, 2020-06-21:
I still have many many errors, even trying the same "make CONFIG=MiniMillgShield" line.

The bit about HAS_HOBBY_SERVO_MOTOR is resolved, but I'm still seeing all the config.h, planner.h, and gpio.h errors.

lllars, 2020-06-22:
Adding #include <iostream> fixes config.h, and #include <cmath> fixes gpio.h

I'm still stumped on the planner.h errors though.

justinclift, 2020-06-22:
Just tried now, and for me it worked.

@lllars Did you run a `git clean -dffx` in the root of the repository directory, to get rid of any files from previous compile attempts?

eg:

```
$ git clean -dffx
$ git pull
$ git submodule update
$ make CONFIG=MiniMillgShield
...
Linking bin/MiniMillgShield-gShield/g2core.elf 
Using linker script: ../Motate/MotateProject/motate/cmsis/TARGET_Atmel/linker_scripts/sam3x/sam3x8/gcc/flash.ld 
--- SIZE INFO ---
   text    data     bss     dec     hex filename
 205556       0   27900  233456   38ff0 bin/MiniMillgShield-gShield/g2core.elf
Making binary bin/MiniMillgShield-gShield/g2core.bin 
Exporting symbols bin/MiniMillgShield-gShield/g2core.elf.txt 
Build 101.03 101.03-533-g490e
```

lllars, 2020-06-22:
@justinclift Thanks, that worked.

Note that I did still run into a couple errors involving settings_default.h when I compiled with `BOARD=gShield` instead of `CONFIG=MiniMillgShield`.  SPINDLE_ACTIVE_HIGH was undeclared and I need to define SPINDLE_SPEED_CHANGE_PER_MS

It's probably too late for today, but I will try this out on my machine tomorrow.

lllars, 2020-06-22:
I've now tried out edge-preview on my machine and am happy to report that the input pin state remains correct following a glitch.  I was able to home the machine without problems.

justinclift, 2020-06-22:
**Whoo Hooo!**

Sounds like good progress. :smile:

willmackey2nd, 2020-10-17:
Current HEAD didn't compile with gShield. Not with default settings nor MiniMillgShield. Took commit https://github.com/synthetos/g2/commit/490eb6cc9a7c55a44b078a564f1126912a56d9fe based on dates in this discussion and it did compile with MiniMill.

Modified default settings with my machines parameters and taking missing stuff from MiniMill until it compiled.

Tested probing with CNCjs + kreso autolevel. Single probing worked fine. Autolevel on the other hand crashes in different ways after 1 or 3 probes. Sometimes it just hangs with `feeder> (AL: probing point 19)
{"sr":{"vel":0,"posz":-1,"mpoz":-1.065}}` in console. Sometimes it crashes Z against bed and continues to push down few seconds, misses some probing point commands during this then continues from some point suddenly.
I also get some `{"err":{"code":204,"msg":"Command rejected by ALARM [$clear to reset]"}}` messages.

Using DUE.

willmackey2nd, 2020-10-17:
Linking some comments about the issue and recent changes here to help anyone googling themselves here.
https://github.com/synthetos/g2/issues/237#issuecomment-711033764
https://github.com/synthetos/g2/issues/237#issuecomment-711060708

willmackey2nd, 2020-10-18:
Did some digging but the source as a whole just looks too complicated to get a quick understanding what happens and where. I was planning to fix this by making sure input state is updated after lockout but no idea where I should be polling that / where's the callback or...

This is a real bummer. I switched from a crappy Chinese controller to G2 in hope of getting auto leveling for PCB milling and kinda took it for granted that stuff like this surely works in G2 and thought a lot of people are using if for this. Reading that this isn't fixable with simple HW filtering, I guess my quick options are to go with GRBL or buy PlanetCNC controller + SW.

giseburt, 2020-10-18:
I am using probing heavily as I use it at least once for every job. Bantam tools and ShopBot are also using it regularly, as they're the ones that collectively found most of the bugs I've mentioned recently.

The only part of that that I don't use and can't say works well is CNCjs with an auto 
-leveling extension. I know we've tested CNCjs and there are people that use it with g2, but in this exact case I can't say how well it works or if the problem exists somewhere in how it's communicating with g2.

If you have a log file or something with a few notes about what's happening at what points along the log we can maybe help sort it out.

justinclift, 2020-10-18:
@willmackey2nd That [kreso autolevel](https://github.com/kreso-t/cncjs-kt-ext) extension for CNCjs looks nifty.  If we can get it working decently well with g2, that'd be useful. :smile:

willmackey2nd, 2020-10-18:
@justinclift yeah it's simple and works and also only alternative for me for now AFAIK. I don't know any better alternatives to use with G2 besides CNCjs. Feel free to slap me in face and point me to a better gcode feeder.

@giseburt  Kreso AL works so that it sends the whole measurement task at once to CNCjs which then starts to execute the gcode. AL then just waits for probe status data, taking notes. Just means that AL itself shouldn't have anything todo with the motion control during probing.

Single probing commands haven't failed once for me but grid probing with AL seems unreliable. I've done now ~30 tests and 
- some times it works fine all the way to the end
- sometimes it's fine except that I can hear maybe 1 skipped step when it's hitting the PCB
- some times it crashes Z hard and does awkward skip of several gcode lines and proceeds, some times it just stops there with probe against the PCB, my multimeter reading 0V at input (= probe tripped), machine not trying to push down, but no errors whatsoever in console. CNCjs just hangs too thinking the job is still being executed.

I don't know if this helps in any way but here's the whole console capture from AL cycle, probing 40x40 area with 7.5mm between probes. This was the kind of fail where it actually told something in console. Somewhere in the middle it crashed Z, continued pushing down and then suddenly proceeded skipping several probe points.
@giseburt I'm happy to dig more info or test other stuff if you want / if it helps you. I did give up for now and ordered MK3Eco as I need to get some stuff done for Christmas but I'd probably have use for g2 anyway later and also I hate giving up / am poor looser.

`{"r":{},"f":[1,0,24]}
feeder> (AL: probing point 7)
feeder> G90 G0 X0.250 Y8.405 Z1
{"qr":46}
feeder> G38.2 Z-1 F50
{"r":{},"f":[1,0,14]}
feeder> G0 Z1
feeder> (AL: probing point 8)
{"qr":45}
feeder> G90 G0 X8.100 Y8.405 Z1
{"sr":{"stat":5,"vel":640.26,"momo":0,"posz":0.90518,"mpoz":-9.81504}}
{"qr":46}
feeder> G38.2 Z-1 F50
{"sr":{"vel":1896.35,"posx":1.04479,"posz":1,"mpox":1.04479,"mpoz":-9.72023}}
{"sr":{"vel":2990.2,"posx":5.90479,"mpox":5.90479}}
{"qr":47}
feeder> G0 Z1
{"sr":{"vel":0.17,"posx":8.10001,"mpox":8.10001}}
{"qr":48}
feeder> (AL: probing point 9)
{"qr":46}
feeder> G90 G0 X15.950 Y8.405 Z1
{"sr":{"stat":7,"vel":50,"feed":50,"momo":1,"posz":0.91901,"mpoz":-9.80122}}
{"sr":{"posz":0.83402,"mpoz":-9.88621}}
{"sr":{"posz":0.74904,"mpoz":-9.97286}}
{"sr":{"posz":0.66405,"mpoz":-10.05618}}
{"sr":{"posz":0.57907,"mpoz":-10.14283}}
{"sr":{"posz":0.49408,"mpoz":-10.22615}}
{"sr":{"posz":0.40909,"mpoz":-10.3128}}
{"sr":{"posz":0.32411,"mpoz":-10.39612}}
{"sr":{"posz":0.23913,"mpoz":-10.48277}}
{"sr":{"posz":0.15414,"mpoz":-10.56609}}
{"sr":{"vel":49.2,"posz":0.06837,"mpoz":-10.65186}}
{"qr":47}
feeder> G38.2 Z-1 F50
{"qr":46}
feeder> G0 Z1
{"sr":{"vel":0.93,"posz":0.0666,"mpoz":-10.65363}}
{"qr":47}
feeder> (AL: probing point 10)
{"qr":48}
feeder> G90 G0 X23.801 Y8.405 Z1
{"prb":{"e":1,"x":8.10000,"y":0.25000,"z":-10.64523,"a":0.00000,"b":0.00000,"c":0.00000}}
{"r":{},"f":[1,0,6]}
feeder> G38.2 Z-1 F50
feeder> G0 Z1
{"qr":46}
feeder> (AL: probing point 11)
{"r":{},"f":[1,0,22]}
feeder> G90 G0 X31.651 Y8.405 Z1
feeder> G38.2 Z-1 F50
{"qr":47}
feeder> G0 Z1
{"r":{},"f":[1,0,25]}
feeder> (AL: probing point 12)
feeder> G90 G0 X39.501 Y8.405 Z1
{"qr":46}
feeder> G38.2 Z-1 F50
[... 932 lines cut ...]
{"sr":{"posz":0.1908,"mpoz":-10.52943}}
{"sr":{"posz":0.10581,"mpoz":-10.61441}}
{"sr":{"posz":0.02083,"mpoz":-10.6994}}
{"sr":{"posz":-0.06416,"mpoz":-10.78438}}
{"sr":{"posz":-0.14914,"mpoz":-10.86937}}
{"sr":{"posz":-0.23413,"mpoz":-10.95435}}
{"sr":{"posz":-0.31911,"mpoz":-11.03934}}
{"sr":{"posz":-0.4041,"mpoz":-11.12432}}
{"sr":{"posz":-0.48908,"mpoz":-11.20931}}
{"sr":{"posz":-0.57407,"mpoz":-11.29429}}
{"sr":{"posz":-0.65905,"mpoz":-11.37928}}
{"qr":47}
{"sr":{"vel":0,"posz":-0.69273,"mpoz":-11.41295}}
{"qr":46}
{"qr":47}
{"qr":48}
{"prb":{"e":1,"x":39.50100,"y":41.02500,"z":-11.41295,"a":0.00000,"b":0.00000,"c":0.00000}}
{"r":{},"f":[1,0,6]}
{"qr":46}
{"sr":{"stat":5,"vel":0.24,"momo":0,"posz":-0.69269,"mpoz":-11.41291}}
{"qr":47}
{"sr":{"vel":995.83,"posz":0.79095,"mpoz":-9.92927}}
{"qr":48}
{"sr":{"stat":3,"vel":0,"posz":1,"mpoz":-9.72023}}
>`