synthetos/g2 #474: Acceleration messed up after G38.2 probing

mhlong10, 2020-07-10:
I've finally root caused an issue which was causing my machine to bang and jerk around.  It doesn't do this when I manually zero Z but does when I use the touch-plate.

**I notice this bug is in both edge and edge-preview.**

_probe_start saves the jerk_max values and _probe_restore_settings restores those values.  The issue is that restore directly sets the values rather than calling cm_set_axis_max_jerk which recalculate max_junction_accel.  This means max_junction_accel doesn't get restored and the machine continues to use the value calculated we set jerk_max was set to jerk_high for the probing cycle.

mhlong10, 2020-07-10:
Submitted PR https://github.com/synthetos/g2/pull/475 to address issue.

justinclift, 2020-07-10:
Sounds like a really good catch. :smile:

justinclift, 2020-07-24:
Closing this, as #475 has been merged.  Thanks heaps @mhlong10. :smile:

closed 2020-07-24 by justinclift