synthetos/g2 #484: Bad state after gpio input action stop

dauriata, 2020-11-03:
With latest edge build 101.03, when triggering a gpio configured as action stop, the motion actually stops but a wrong position is reported ( updated to the target position )
example moving from y0 to y200 and hitting stop in the middle of move : 

[image]

As a side note, I was expected GPIO action stop and fast stop to behave like ! feedhold, but looking at the code I see that "fast stop" is not implemented and is like "stop". And "stop" does a program stop, not just like "!" feedhold. I think it would be more useful to have a hardware input button for just feedhold rather than feedhold followed by program stop. in my opinion program stop is more a function that would be done from the higher interface used to control the job.

dauriata, 2020-11-03:
Just find out that this might be same bug as #479. Although this case is reproducible.

lutorm, 2021-07-05:
I would also like to be able to trigger a feed hold with a GPIO.