synthetos/g2 #487: edge-preview: missing acceleration/deceleration when sending multiple moves at once

prof7bit, 2021-01-07:
I was just playing around with the edge-preview branch (to check whether my other bug was fixed there [it is not]) and found the following odd behavior:

If I send something like this:

g1 z100 f10000
g1 z0 f10000
g1 z100 f10000
g1 z0 f10000
g1 z100 f10000
g1 z0 f10000
g1 z100 f10000
g1 z0 f10000

Then it will not decelerate/accelerate between the direction reversals, it will just hard switch direction at full speed, resulting in severe mechanical stress and loud bangs as if it crashed into something!

The edge branch does not have this bug.

I probably won't be available to test the fix on the preview branch (should it ever be fixed at all) because I am now moving back to the half finished edge branch which does not have this error. But I wanted to report it anyways, so it can be worked on in case this project ever becomes actively developed again.

giseburt, 2021-01-07:
This indicates that your `jt` (`JUNCTION_INTEGRATION_TIME` in settings) value is too high. It basically tells the system how aggressive of cornering your machine can handle.

The test you have is great, except you likely want to have the same test for X and Y and combined XY as well. I'd start with setting `jt` to something low, like `0.5`, and raising it slowly until that axis stalls. Verify that value is good for the other axes.

Note that `jt` works with your axis jerk settings (`xjm`, or `X_JERK_MAX`, and `yjm`, etc.), so if one axis is having trouble with the `jt` that the other axis are fine with, then lower that axis' jerk max setting.

-Rob

mhlong10, 2021-01-07:
If you ran a probe cycle prior to this then it might be caused by https://github.com/synthetos/g2/pull/475 which I don't believe  wasn't brought forward into edge-preview.