synthetos/g2 #492: Spiral roughing hiccup?

Mgilbride, 2021-03-28:
Not sure this is an issue. 

I have run some motion tests with a new cnc router build running g2 on a DUE. Gcode is from RhinoCam. I am running firmware version 100.26. 

In one of my tests I am using spiral roughing. At the "same spot" on each pass of the spiral the machine seems to jump back a mm or two causing a noticeable jolt in otherwise smooth motion. I do not see any similar behavior when executing this as a parallel roughing operation. I still need to try this as a circular offset rather than spiral. 

I see no indication that this is mechanical. Any thoughts if this could be a g2 vs RhinoCam issue? I am suspecting some small, hard to see issue with arc interpolation. Not sure how to troubleshoot this behavior and could use some suggestions.

ril3y, 2021-03-28:
Can you attach the gcode?

Mgilbride, 2021-03-28:
Absolutely. Attached here. 

Many thanks.

[TestGeometry2.txt](https://github.com/synthetos/g2/files/6217493/TestGeometry2.txt)

Mgilbride, 2021-03-28:
I need to do more testing. I noticed this page of the post processor. It was set to helical interpolation. There's an option for spiral interpolation. 

[image]

Mgilbride, 2021-03-28:
OK. After another couple of hours testing... I updated the helix/spiral arc interpolation settings in the postprocessor. I utilized arc fitting in the machining settings. All pictured in screenshots here. Updated Gcode also attached here. I get less regularity and severity in the pauses / jumps. I caught an example on a short video. They happen the same place on each layer. But they now don't occur every pass. So, I think this is down to arc interpolation. And it could just be that I am not knowledgeable enough about the settings here. And / or there is a limitation in G2 firmware or RhinoCam.

https://user-images.githubusercontent.com/741811/112767131-f2927f00-8fe2-11eb-8606-676ac2718e38.MOV
[image]
[image]
[image]
[image]
[TestGeometry8.txt](https://github.com/synthetos/g2/files/6218012/TestGeometry8.txt)

closed 2023-02-16 by ril3y