synthetos/g2 #494: Skew correction ideas?

Mgilbride, 2021-04-18:
After completing a couple CNC builds I (and the rest of the diy cnc machine building community)  keep tangling with the finer points of squaring the machine during final setup. Marlin includes some firmware based tools for skew correction that work with varying success. Others have written their own post processor tools that rewrite a gcode file to correct for observed skew on a machine. General concensus is solving skew problems by adjusting the mechanical details of the machine are the correct way to go. But, In my experience it can be quite difficult to dial in larger, heavier machines unless this final squaring process was carefully considered and designed for (such as adjustment bolts that allow for fine nudging of frame components during squaring). And, even with these provisions I would still expect some headaches dialing in and maintaining squareness of the gantry. 

Dual Y motors help tackle this to some degree. But, this solution is limited to a subset of machines and I don't think linear guides like hiwin rails and blocks are very tolerant of this mode of gantry squaring. In the end, the problem still boils down to trying to iron out small values of skew observed during squareness tests. 

I don't see much discussion of these details here in the wiki. Have any of you tackled fine adjustments to machine alignment through firmware settings? Is there any merit to adding Marlin-like functionality to G2 firmware?  Thoughts and shared experience here would be appreciated.

giseburt, 2021-04-19:
I really like this idea. I've only idly thought of it while mechanically squaring a machine, generally dismissing it for fear that a mechanical skew would cause additional wear and other compound issues.

But I see that that's not necessary the case.

So there is already a matrix transform on every endpoint run into the system. A consistent skew across the coordinate system (vs a non-linear one) would be easily handled by that, and would only require a means to configure it for the skew.

The matrix is currently only configurable via the "tram" command, and it applies a rotation to place the last three probe points on `Z0`. A similar command and similar configuration could be used, and no additional work should be needed.

jkuusama, 2021-04-19:
This is fairly trivial, at least on surface. Once the correction parameter c is known, the software moves the machine a little in X direction when it does an Y move. To go to (X, Y), the machine actually goes to (X+c*Y, Y). The complication comes in the user interface, as this needs to be hidden from the user; there needs to be two sets of coordinates: Where the machine actually is, and what is shown to the user.

ril3y, 2021-11-05:
Closing for lack of activity.  You can re-open if you wish.

closed 2021-11-05 by ril3y