synthetos/g2 #498: Spindle sometimes does not enable

lutorm, 2021-05-16:
Hey all,

I'm having a problem where my spindle stops working. I leave my Due up but turn off the motor power supply and suspend the computer. Then when I reconnect, re-power the motors (which also powers the output driver to the spindle), the spindle no longer turns on when I do an M3. I've verified that there's no action on the Due spindle enable GPIO. Motion works, so the controller hasn't panicked, and if I do $sp it says the spindle is enabled. Any ideas about what could cause this and how to troubleshoot?

discoordination, 2021-07-22:
I have found that the software doesn't like it if the computer goes to sleep which seems to be related to the usb hub power management.  Now with the machine at my workplace we have attached it to a pc there. and what I have done is to right click all the usb root hubs in device manager and go to properties -> power management  and disable 'allow the pc to turn off this device to save power'.  I have noticed that now the CNC works again in the morning after the pc has gone to sleep all night, much more reliably.

justinclift, 2021-07-25:
Interesting, that's a really good data point @discoordination. :smile:

ril3y, 2021-11-05:
Closing for lack of activity.  You can re-open if you wish.

closed 2021-11-05 by ril3y