synthetos/g2 #504: Arduino Due with Edge-preview+gShield not working ///'laser_tool' was not declared in this scope///

martonpetroczki93, 2021-12-27:
I tried to upload the new version but it didn't work. I use an arduino due board with my own shield and the old version worked fine.
This error message appears:
[image]
I don't understand why the laser is mixed in and what is this? "#warning SPINDLE_ENABLE_OUTPUT_NUMBER is defaulted to 4!"
I found this, but gShield doesn't use it.
[image]

(sorry, my English is bad)

justinclift, 2021-12-28:
> the old version worked fine.

When you say "the old version", are you meaning an older board, or an older file for uploading, or something else? :smile:

My guess is that you've copied one of the existing settings files ([here](https://github.com/synthetos/g2/tree/edge-preview/g2core/settings)) to `cheapo_laser.h`, and the new one isn't working but the old one is?

> sorry, my English is bad

So far, it's easily good enough. :smile:

martonpetroczki93, 2021-12-28:
I didn't say it exactly.  So the old version is the /master/. I completely deleted the old g2 master folder and I downloaded the edge-preview. What am I doing wrong? :(

nickbooties, 2021-12-28:
Yeah, looks like  https://github.com/synthetos/g2/commit/423c83f233f1ff414521b9f66170e530d95ad14c broke a bunch of stuff for Due/gBoard users. I reverted that and it compiled no problems.

martonpetroczki93, 2022-01-02:
Thanks for your help, it solved the problem.

closed 2022-01-02 by martonpetroczki93