synthetos/g2 #512: Compile g2core for gShield on Windows

Element21302, 2022-08-04:
I'm still very new to C code and I was having issues compiling g2core for gShield. I was able to get it to compile after adding this to the **settings_shapeoko2.h** file (copied from **settings_cheapo_laser.h**):

```
#define HAS_LASER                   0 
#define HAS_PRESSURE                0 
#define LASER_TOOL                  1   // default tool is 1 - note that TOOLS may be limited to 5!
#define LASER_MIN_S                 0.0001// {th2mns:0.0001}
#define LASER_MAX_S                 255.0 // {th2mxs:255}
#define LASER_MIN_PPM               200   // {th2mnp:200}
#define LASER_MAX_PPM               8000  // {th2mxp:8000}
#define LASER_PULSE_DURATION        100                      // in microseconds {th2pd:5}
```

I also need to comment out a few of the lines in **0_hardware.cpp** to ignore **laser_tool**
This seems like the wrong approach, but I don't really know what I'm doing. I can get the other configurations to compile without modification. 

A few things I'm curious about are:

1. Should I need to define laser settings for gShield? 
2. Does it always load the **settings_default.h** file?
3. How can I create my own settings file and include that instead.

Thanks to anyone that has any information that can help me. This is all new and the documentation is not

RobMackie, 2022-08-23:
I'm also having trouble building the DUE target. I'm wondering if I don't know the correct incantation of "make" to get all the right platform things figured out. Since the wiki now says that the DUE is the default target, I tried just doing:
>  make

But you need a board. So I tried:
make PLATFORM=DUE BOARD=gShield
and 
make BOARD=gShield    #depends on due being the default target platform

The last two built a lot of files but ended in what look like the same errors you must have been hitting (based on what you say you changed.)

Anyone out there who knows why DUE doesn't build - including that we're just doing it wrong?

Seems like if anything would have an unbroken build, it would be the "default" target. I'm pretty sure I built it in Jan of 2021. I even started a document on how to do it under docker for desktop on windows (and then someone dressed that up and made it more readable)  

I'm going to start looking to see if I can determine what has changed.  (because I, too, would like to deploy it on due).

Element21302, 2022-08-24:
I was able to compile and flash after making the changes, and it seems to run just fine.

ril3y, 2023-02-16:
make PLATFORM=DUE BOARD=gShield
This is the correct make command!  Thanks for fixing that.

closed 2023-02-16 by ril3y