synthetos/g2 #514: Which version?

simonemarin, 2023-02-16:
Hallo been using g2(core) some years ago and all well, now I am a bit confused about versioning. Coming from https://github.com/synthetos/g2/issues/513 when @ril3y says "The default branch in g2core is edge-preview", then if I get the latest Release I get version 101.03, if I git clone edge-preview, which version is that? Isn t any bin available? Tried to compile in WSL in Windows but got stuck at some point with MAKE and before trying to debug that, since I see I am not the only one, wanted to get some clarification about versions and possible compiled bins out there, thanks.
Simone

ril3y, 2023-02-16:
For now it is a bit confusing.  We are working on cleaning things up a bit.  We added WSL support this morning.  You will need to update the motate submodule and pull the latest of g2core.
I created a binary that you can try out.  I would use with caution as I have not tested it at all.  Might work might not,  I don't have a due setup to test on at the moment.

https://drive.google.com/file/d/1SAU-vpRx9YbPC1ikLeY-4K2qexnzPPKU/view?usp=sharing

you can get it there.

We will continue to clean up the code base and make the due more approachable to users.

simonemarin, 2023-02-16:
I am getting the same issue as mentioned before:
```
CNCjs 1.10.1 [TinyG]
Connected to COM32 with a baud rate of 115200
feeder> {ej:1}
feeder> {jv:4}
feeder> {qv:1}
feeder> {sv:1}
feeder> {si:100}
feeder> {spe:n}
feeder> {spd:n}
feeder> {spc:n}
feeder> {sps:n}
feeder> {com:n}
feeder> {cof:n}
feeder> {sr:{stat:t,line:t,vel:t,feed:t,unit:t,coor:t,momo:t,plan:t,path:t,dist:t,frmo:t,tool:t,posx:t,posy:t,posz:t,posa:t,posb:t,posc:t,mpox:t,mpoy:t,mpoz:t,mpoa:t,mpob:t,mpoc:t,spe:t,spd:t,spc:t,sps:t,com:t,cof:t}}
feeder> {sys:n}
feeder> {mt:n}
feeder> {pwr:n}
feeder> {qr:n}
feeder> {sr:n}
```

ril3y, 2023-02-16:
Try using coolterm and sending that single line. See how it responds

ril3y, 2023-02-16:
Ok this is something to do with CNCjs.  As I can replicate your issue.

Here is how it works.  
Open cncjs try to connect.  Nothing happens.  (however g2 is not in any error and is just fine)
Then close the port.

Open Coolterm connect.  It immediately displays g2 sending the 
{"r":{"fv":0.99,"fb":101.06,"fbs":"101.03-559-g28e7-dirty","fbc":"settings_default.h","hp":"gQuintic","hv":"QUINTIC_REVISION","id":"5337-3100-3832-5335-3330-3431-3635-303","msg":"SYSTEM READY"},"f":[1,0,1]}
Message.

I suspect cncjs is not opening the port ?  @MitchBradley do you have any ideas about this?

ril3y, 2023-02-16:
Oh I will say that once the initial connection is made with cool term and then closed.  CNCjs works ok after that.

ril3y, 2023-02-16:
@giseburt Do you have any input on this?

MitchBradley, 2023-02-16:
I stopped supporting CNCjs a few months ago, after several years of frustration...

ril3y, 2023-02-16:
> I stopped supporting CNCjs a few months ago, after several years of frustration...

Not exactly the input I was after :)  Just kidding!

simonemarin, 2023-02-16:
Hallo, yes I can ask for $fv in the Arduino serial monitor then CNCJS still fails, then which software would you advice to use with G2core?

MitchBradley, 2023-02-16:
I think that UGS supports g2core now.

ril3y, 2023-02-16:
You can open a connection to coolterm or arduino serial monitor etc.  Then connect to cncjs.  Or you can UGS.  I can confirm UGS opens the connection correctly.  That being said you should open a issue on cncjs github and link this issue. 

I am going to close this for now.

closed 2023-02-16 by ril3y

simonemarin, 2023-02-16:
UGS sems to work, if it is ok I will report here my findings.

ril3y, 2023-02-16:
I think you should still open a issue on cncjs.

simonemarin, 2023-02-16:
https://github.com/cncjs/cncjs/issues/812

@ril3y can I follow the pinout for Due as described in https://github.com/synthetos/g2/wiki/Arduino-DUE-Pinout-for-g2core ? Or can you provide the pinout you used for compiling? My Z axis does not move.

ril3y, 2023-02-16:
@simonemarin 
https://github.com/synthetos/g2/blob/edge-preview/g2core/board/ArduinoDue/gShield-pinout.h#L131-L159

I think this is the code you are looking for.