synthetos/g2 #516: Project status

timonsku, 2024-01-03:
Hi,
I was wondering what the status of the project is and if some of the hardware that was hinted at years ago will see the light of day.
It seems some commercial vendors actively use g2 (Bantam) even in new products. Has development shifted to a semi closed source model there?

Similiar question for TinyG v9 hardware which seems to have shipped for commercial use but I can't really find any schematics or details on that.

breiler, 2024-02-13:
@aldenhart, @ril3y, @giseburt - the synthentos web page is currently down: https://synthetos.com/.

I know that there hasn't been much development here lately, but I am getting some questions about a couple of features for which I need the hardware specs/pinout for your gQuintic board. Is it possible to add those to the wiki?

ril3y, 2024-02-13:
I am looking in to the site issue now.  Can you tell me more about what you
are doing with g2?  Also you can edit the wiki that is fine.  What is it
you are trying to update?

Riley

breiler, 2024-02-14:
There are a couple of users trying to use probing with g2core and UGS which is not yet implemented. To be able to implement this I need diagrams and more documentation of the gQuintic board which I was hoping to be available on the web page. 

I can see that you got the web page up and running, but there isn't much information about the board so I'll have to figure this out some other way.

ril3y, 2024-02-14:
https://github.com/synthetos/g2/wiki

Does not have what you are after?  What specifically are you wanting to
know.  Our gQuintic did not ever get sold.  We have lots of demo boards.
Diagrams for what?  I can try to provide data.

When you say probing for like pcbs to make up for the offset in material?

Riley

breiler, 2024-02-14:
Sorry, this got really out of topic...

> When you say probing for like pcbs to make up for the offset in material?

Yes, both auto leveling and for Z/XY/XYZ/Hole-probing to locate the material.

> Our gQuintic did not ever get sold.

I got a gQuintic board from you guys to develop against a long time ago. But in order to configure probing, I need to know the IO numbering for the inputs. So without the schematics it will be a bit difficult to setup. There is an empty placeholder for it here which would have been useful:
https://github.com/synthetos/g2/wiki/gQuintic-Specs#io-documentation

I have an Arduino Due with a G2core shield, but it will take a lot of time to get it wired up to my test machine.